moveit/moveit2 #2795: Humble: MoveIt2 Getting Started Tutorial - colcon build failed(moveit_kinematics)

fabriros, 2024-04-16:
### Description
Not able to build the Colcon workspace, even if i'm following the tutorial at [https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html].

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build, (I've tried both main and humble)

### Steps to reproduce
Follow the tutorial [https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html]

### Expected behaviour
Build colcon workspace without error.

### Actual behaviour
`Summary: 35 packages finished [17min 19s]
  1 package failed: moveit_kinematics
  5 packages aborted: moveit_planners_ompl moveit_ros_perception moveit_ros_robot_interaction moveit_ros_warehouse moveit_visual_tools
  2 packages had stderr output: moveit_kinematics moveit_visual_tools
  28 packages not processed
`

### Backtrace or Console output
`Starting >>> moveit_common
Starting >>> moveit_msgs
Starting >>> moveit_configs_utils
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> robotiq_description
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support
Starting >>> kortex_api
Starting >>> rviz_visual_tools
Starting >>> serial
Starting >>> robotiq_controllers
Finished <<< moveit_configs_utils [4.22s]
Finished <<< moveit_common [4.33s]
Starting >>> rviz_marker_tools
Finished <<< moveit_resources_fanuc_description [4.40s]
Finished <<< moveit_resources_panda_description [4.41s]
Starting >>> moveit_resources_panda_moveit_config
Starting >>> moveit_resources_fanuc_moveit_config
Starting >>> dual_arm_panda_moveit_config
Finished <<< moveit_resources_prbt_support [4.43s]
Finished <<< robotiq_description [4.45s]
Starting >>> kortex_description
Finished <<< moveit_resources_pr2_description [4.49s]
Finished <<< dual_arm_panda_moveit_config [1.38s]
Finished <<< kortex_api [5.84s]
Finished <<< moveit_resources_fanuc_moveit_config [1.46s]
Starting >>> kortex_driver
Finished <<< moveit_resources_panda_moveit_config [1.49s]
Starting >>> moveit_resources
Finished <<< kortex_description [1.71s]
Finished <<< moveit_resources [1.18s]
Finished <<< serial [14.8s]
Starting >>> robotiq_driver
Finished <<< robotiq_controllers [26.2s]
Finished <<< rviz_marker_tools [28.5s]
Finished <<< kortex_driver [33.0s]
Starting >>> kortex_bringup
Finished <<< kortex_bringup [1.91s]
Finished <<< robotiq_driver [52.8s]
Starting >>> robotiq_hardware_tests
Finished <<< robotiq_hardware_tests [10.3s]
[Processing: moveit_msgs, rviz_visual_tools]
Finished <<< rviz_visual_tools [2min 18s]
[Processing: moveit_msgs]
Finished <<< moveit_msgs [2min 51s]
Starting >>> moveit_core
Starting >>> moveit_task_constructor_msgs
[Processing: moveit_core, moveit_task_constructor_msgs]
Finished <<< moveit_task_constructor_msgs [40.5s]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
[Processing: moveit_core]
Finished <<< moveit_core [5min 0s]
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_planners_stomp
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pick_ik
Starting >>> pilz_industrial_motion_planner_testutils
[Processing: chomp_motion_planner, moveit_planners_stomp, moveit_resources_prbt_ikfast_manipulator_plugin, moveit_ros_occupancy_map_monitor, moveit_simple_controller_manager, pick_ik, pilz_industrial_motion_planner_testutils]
Finished <<< moveit_simple_controller_manager [55.8s]
Starting >>> moveit_plugins
Starting >>> moveit_ros_control_interface
Finished <<< moveit_plugins [2.58s]
Finished <<< pilz_industrial_motion_planner_testutils [1min 6s]
Finished <<< chomp_motion_planner [1min 18s]
Starting >>> moveit_planners_chomp
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [1min 26s]
Finished <<< moveit_ros_occupancy_map_monitor [1min 38s]
Starting >>> moveit_ros_planning
Finished <<< moveit_planners_stomp [1min 56s]
Finished <<< moveit_ros_control_interface [1min 14s]
Finished <<< moveit_planners_chomp [59.5s]
[Processing: moveit_ros_planning, pick_ik]
[Processing: moveit_ros_planning, pick_ik]
Finished <<< pick_ik [3min 19s]
[Processing: moveit_ros_planning]
[Processing: moveit_ros_planning]
[Processing: moveit_ros_planning]
[Processing: moveit_ros_planning]
Finished <<< moveit_ros_planning [4min 7s]
Starting >>> moveit_kinematics
Starting >>> moveit_ros_warehouse
Starting >>> moveit_ros_robot_interaction
Starting >>> moveit_planners_ompl
Starting >>> moveit_ros_perception
Starting >>> moveit_visual_tools
[Processing: moveit_kinematics, moveit_planners_ompl, moveit_ros_perception, moveit_ros_robot_interaction, moveit_ros_warehouse, moveit_visual_tools]
[Processing: moveit_kinematics, moveit_planners_ompl, moveit_ros_perception, moveit_ros_robot_interaction, moveit_ros_warehouse, moveit_visual_tools]
--- stderr: moveit_kinematics
In file included from /home/fabrizio/ws_moveit/src/moveit2/moveit_kinematics/kdl_kinematics_plugin/include/moveit/kdl_kinematics_plugin/kdl_kinematics_plugin.h:42,
                 from /home/fabrizio/ws_moveit/src/moveit2/moveit_kinematics/kdl_kinematics_plugin/src/kdl_kinematics_plugin.cpp:37:
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::refresh_dynamic_parameters()’:
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:146:36: error: expected unqualified-id before ‘.’ token
  146 |       auto& entry = updated_params..joints_map[value_1];
      |                                    ^
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘rcl_interfaces::msg::SetParametersResult kdl_kinematics::ParamListener::update(const std::vector<rclcpp::Parameter>&)’:
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:217:32: error: expected unqualified-id before ‘.’ token
  217 |                 updated_params..joints_map[value_1].weight = param.as_double();
      |                                ^
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp: In member function ‘void kdl_kinematics::ParamListener::declare_params()’:
/home/fabrizio/ws_moveit/build/moveit_kinematics/kdl_kinematics_plugin/kdl_kinematics_parameters/include/kdl_kinematics_parameters.hpp:308:36: error: expected unqualified-id before ‘.’ token
  308 |       auto& entry = updated_params..joints_map[value_1];
      |                                    ^
gmake[2]: *** [kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/build.make:76: kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/src/kdl_kinematics_plugin.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:363: kdl_kinematics_plugin/CMakeFiles/moveit_kdl_kinematics_plugin.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_kinematics [1min 26s, exited with code 2]
Aborted  <<< moveit_ros_robot_interaction [1min 53s]
Aborted  <<< moveit_ros_perception [2min 32s]
Aborted  <<< moveit_visual_tools [2min 49s]
Aborted  <<< moveit_ros_warehouse [3min 35s]
Aborted  <<< moveit_planners_ompl [3min 42s]

Summary: 35 packages finished [17min 19s]
  1 package failed: moveit_kinematics
  5 packages aborted: moveit_planners_ompl moveit_ros_perception moveit_ros_robot_interaction moveit_ros_warehouse moveit_visual_tools
  2 packages had stderr output: moveit_kinematics moveit_visual_tools
  28 packages not processed
`

Thanks in advance.

jay-waves, 2024-04-17:
According to issues #2749 and #2758, the problem can be attributed to changes in the dependency [generate_parameter_library](https://github.com/PickNikRobotics/generate_parameter_library). This error has been fixed in the latest version of that library, although an older version may still be installed through rosdep.

I simply removed all the redundant commas to compile successfully. However, it is recommended that you follow @kimsooyoung's solution to compile the latest version of generate_parameter_library from source in your ROS overlay. For more details, see [this comment](https://github.com/ros-planning/moveit2/issues/2749#issuecomment-2044886248).

fabriros, 2024-04-17:
> According to issues #2749 and #2758, the problem can be attributed to changes in the dependency [generate_parameter_library](https://github.com/PickNikRobotics/generate_parameter_library). This [...]

Thank you very much.
I've done what you have suggested ( @kimsooyoung's solution). But now there are other issues.
In particular following the [tutorials](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) can't go forward.
[image]
[image]
[image]

kimsooyoung, 2024-04-17:
Could you delete whole ws legacies and clone again? Seems like your pkgs aren't looking forward to the right version.
For instance, below codes are the original humle version `execute_task_solution_capability.cpp`.
in your second screenshot,

[checkout code permalink](https://github.com/ros-planning/moveit_task_constructor/blob/6b0f2c872f68595637e54935e083f15d32924e16/capabilities/src/execute_task_solution_capability.cpp#L143)

```c++
	plan.plan_components_.reserve(solution.sub_trajectory.size());
	for (size_t i = 0; i < solution.sub_trajectory.size(); ++i) {
		const moveit_task_constructor_msgs::msg::SubTrajectory& sub_traj = solution.sub_trajectory[i];

		plan.plan_components_.emplace_back();
		plan_execution::ExecutableTrajectory& exec_traj = plan.plan_components_.back();

		// define individual variable for use in closure below
		const std::string description = std::to_string(i + 1) + "/" + std::to_string(solution.sub_trajectory.size());
		exec_traj.description_ = description;
```

However, As you see in your second screenshot, your codes are using `plan.plan_components_` not the `plan.plan_components`
So I think you have to check pkg versions first.

closed 2024-07-22 by github-actions[bot]