moveit/moveit2 #2797: /clear_octomap service does not update the octomap properly in MoveIt2

Danilrivero, 2024-04-18:
### Description

Directly related to the following solved issues from MoveIt.

https://github.com/ros-planning/moveit/issues/3380
https://github.com/ros-planning/moveit/pull/3134

When the pcd topic does not publish, the "clear octomap" option in MotionPlanning from rviz2 will not clear the map. If the pcd topic does publish, the map will update based on the pcd and can't be cleared completely. In the move_group's PS, the octomap is correctly cleared even though it seems not to update through rviz2. @rhaschke may have information on this looking at his PR in MoveIt.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary
* If binary, which release version? 2.5.5-1
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone DDS

### Steps to reproduce
Publish a pointcloud using the _sensors_3d.yaml_ module for the generation of the octomap. The generated octomap will not clear entirely unless updated at the same time through the publishing of a new pcd (moreover always having a PS containing octomap parts).

### Expected behaviour
When calling the "/clear_octomap" service the module should clear the octomap even though it would presumably update within the next update if pcd is being published.

### Actual behaviour
The module does not update the octomap unless being updated with the publishing of a new pcd.

[image]

### Backtrace or Console output

Usage of the following configuration file: [sensors_3d.yaml](https://gist.github.com/Danilrivero/6776b24bf832fc138c68c1d0b5730e95).

Seems like it requires to sync the PR made in MoveIt to MoveIt2.

patrickKXMD, 2024-04-23:
Hi, what do you use for planning? `MoveItCpp` in your node or a standalone `move_group` node?

The button in rviz plugin triggers `/clear_octomap` service which is a capacity in `move_group` node.
If you are using `MoveItCpp` inside your own node, use `moveit_cpp->getPlanningSceneMonitorNonConst()->clearOctomap();`to clear the octomap.

Danilrivero, 2024-04-23:
> Hi, what do you use for planning? `MoveItCpp` in your node or a standalone `move_group` node?
>
> The button in rviz plugin triggers `/clear_octomap` service which is a capacity in `move_group` node. [...]

In this case I use the standalone `move_group` node. I have verified that by constantly updating the octomap, this is, constantly publish a pointcloud to the defined topic inside the` sensors_3d.yaml` it will properly clear due to the octomap constantly being updated.

Danilrivero, 2024-06-10:
Bumping this in case anyone is available to do this backport since the fixes exist back in MoveIt1. Else I will do it once I am available.

closed 2024-10-09 by github-actions[bot]