moveit/moveit2 #2806: TM5-900 "No kinematics plugins defined." and "Failed to call service get_planning_scene"

AlbertLord, 2024-04-24:
## Introduction
Hello,
I'm trying to make the TM ROS2 Driver ([tmr_ros2](https://github.com/TechmanRobotInc/tmr_ros2/tree/humble)) work with the main brach MoveIt2, because I intend to use moveit_py later. I made some alterations to the source material, but I'm not familiar enough with ROS2 to now what I'm doing exactly.

## Enviroment

- Ubuntu 22.04 on WSL
- ROS2 Humble - Binary
- MoveIt2 - Source from main branch
- TM ROS2 Driver - Source with moveit packages from Foxy branch

## This is how I got here:
I'm using Ubuntu 22.04 and installed ROS2 Humble. Then I followed the Moveit2 guide to bulid it from the main branch. Aside from a couple of type casting warnings, everything went smoothly. Since the TM ROS2 Driver's [Humble branch](https://github.com/TechmanRobotInc/tmr_ros2/tree/humble) doesn't contain the Moveit2 packages I had to copy it from the [Foxy branch](https://github.com/TechmanRobotInc/tmr_ros2/tree/foxy) (a month ago I used the driver just like this without any issue but I could only use Cartesian Planning in RViz). Building the driver with the source built MoveIt2 :

```
--- stderr: tm_moveit_cpp_demo                               
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:134:11: error: ‘planning_interface’ in namespace ‘moveit’ does not name a type; did you mean ‘planning_pipeline_interfaces’?
  134 |   moveit::planning_interface::MoveItCppPtr moveit_cpp_;
      |           ^~~~~~~~~~~~~~~~~~
      |           planning_pipeline_interfaces
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp: In member function ‘void MoveItCppDemo::run()’:
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:67:5: error: ‘moveit_cpp_’ was not declared in this scope; did you mean ‘moveit_cpp’?
   67 |     moveit_cpp_ = std::make_shared<moveit::planning_interface::MoveItCpp>(node_);
      |     ^~~~~~~~~~~
      |     moveit_cpp
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:67:44: error: ‘planning_interface’ is not a member of ‘moveit’; did you mean ‘planning_pipeline_interfaces’?
   67 |     moveit_cpp_ = std::make_shared<moveit::planning_interface::MoveItCpp>(node_);
      |                                            ^~~~~~~~~~~~~~~~~~
      |                                            planning_pipeline_interfaces
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:67:24: error: parse error in template argument list
   67 |     moveit_cpp_ = std::make_shared<moveit::planning_interface::MoveItCpp>(node_);
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:67:74: error: no matching function for call to ‘make_shared<<expression error> >(rclcpp::Node::SharedPtr&)’
   67 |     moveit_cpp_ = std::make_shared<moveit::planning_interface::MoveItCpp>(node_);
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/c++/11/memory:77,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:153,
                 from /home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:46:
/usr/include/c++/11/bits/shared_ptr.h:875:5: note: candidate: ‘template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)’
  875 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
/usr/include/c++/11/bits/shared_ptr.h:875:5: note:   template argument deduction/substitution failed:
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:67:74: error: template argument 1 is invalid
   67 |     moveit_cpp_ = std::make_shared<moveit::planning_interface::MoveItCpp>(node_);
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:72:13: error: ‘moveit::planning_interface’ has not been declared
   72 |     moveit::planning_interface::PlanningComponent arm("tmr_arm", moveit_cpp_);
      |             ^~~~~~~~~~~~~~~~~~
/home/andrasmakany/tmdriver_ws/src/tm_moveit_cpp_demo/src/run_moveit_cpp.cpp:103:5: error: ‘arm’ was not declared in this scope
  103 |     arm.setGoal("ready1");
      |     ^~~
gmake[2]: *** [CMakeFiles/run_moveit_cpp.dir/build.make:76: CMakeFiles/run_moveit_cpp.dir/src/run_moveit_cpp.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/run_moveit_cpp.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< tm_moveit_cpp_demo [9.20s, exited with code 2]

Summary: 17 packages finished [9.43s]
  1 package failed: tm_moveit_cpp_demo
  1 package had stderr output: tm_moveit_cpp_demo
```

As best as I could I corrected these until the package built successfully. The end result was:
```
#include <thread>
#include <rclcpp/rclcpp.hpp>
#include <moveit/moveit_cpp/moveit_cpp.h>
#include <moveit/moveit_cpp/planning_component.h>
#include <moveit/robot_state/conversions.h>
#include <moveit_msgs/msg/display_robot_state.hpp>
[... 169 lines cut ...]
[rviz2-4] [INFO] [1713920514.589422218] [rviz2.moveit.core.robot_model]: Loading robot model 'tm5-900'...
[rviz2-4] [WARN] [1713920514.622158312] [rviz2.moveit.ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[run_moveit_cpp-5] [INFO] [1713920514.650695221] [moveit_3730791688.moveit.ros.planning_pipeline]: Successfully loaded planner 'OMPL'
[run_moveit_cpp-5] [INFO] [1713920514.653099982] [run_moveit_cpp]: Try loading adapter 'default_planning_request_adapters/ResolveConstraintFrames'
[run_moveit_cpp-5] [INFO] [1713920514.655213001] [run_moveit_cpp]: Loaded adapter 'default_planning_request_adapters/ResolveConstraintFrames'
[run_moveit_cpp-5] [INFO] [1713920514.655310477] [run_moveit_cpp]: Try loading adapter 'default_planning_request_adapters/ValidateWorkspaceBounds'
[run_moveit_cpp-5] [INFO] [1713920514.656588259] [run_moveit_cpp]: Loaded adapter 'default_planning_request_adapters/ValidateWorkspaceBounds'
[run_moveit_cpp-5] [INFO] [1713920514.656668648] [run_moveit_cpp]: Try loading adapter 'default_planning_request_adapters/CheckStartStateBounds'
[run_moveit_cpp-5] [INFO] [1713920514.656936921] [run_moveit_cpp]: Loaded adapter 'default_planning_request_adapters/CheckStartStateBounds'
[run_moveit_cpp-5] [INFO] [1713920514.656970915] [run_moveit_cpp]: Try loading adapter 'default_planning_request_adapters/CheckStartStateCollision'
[run_moveit_cpp-5] [INFO] [1713920514.657003259] [run_moveit_cpp]: Loaded adapter 'default_planning_request_adapters/CheckStartStateCollision'
[run_moveit_cpp-5] [INFO] [1713920514.657017623] [run_moveit_cpp]: Try loading adapter 'default_planning_request_adapters/AddTimeOptimalParameterization'
[run_moveit_cpp-5] [FATAL] [1713920514.657405497] [run_moveit_cpp]: Exception while loading planning adapter plugin 'default_planning_request_adapters/AddTimeOptimalParameterization': According to the loaded plugin descriptions the class default_planning_request_adapters/AddTimeOptimalParameterization with base class type planning_interface::PlanningRequestAdapter does not exist. Declared types are  default_planning_request_adapters/CheckForStackedConstraints default_planning_request_adapters/CheckStartStateBounds default_planning_request_adapters/CheckStartStateCollision default_planning_request_adapters/ResolveConstraintFrames default_planning_request_adapters/ValidateWorkspaceBounds planning_pipeline_test/AlwaysSuccessRequestAdapter
[run_moveit_cpp-5] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[run_moveit_cpp-5]   what():  According to the loaded plugin descriptions the class default_planning_request_adapters/AddTimeOptimalParameterization with base class type planning_interface::PlanningRequestAdapter does not exist. Declared types are  default_planning_request_adapters/CheckForStackedConstraints default_planning_request_adapters/CheckStartStateBounds default_planning_request_adapters/CheckStartStateCollision default_planning_request_adapters/ResolveConstraintFrames default_planning_request_adapters/ValidateWorkspaceBounds planning_pipeline_test/AlwaysSuccessRequestAdapter
[ERROR] [run_moveit_cpp-5]: process has died [pid 202198, exit code -6, cmd '/home/andrasmakany/tmdriver_ws/install/tm_moveit_cpp_demo/lib/tm_moveit_cpp_demo/run_moveit_cpp --ros-args --params-file /home/andrasmakany/tmdriver_ws/install/ [... cut ...]
[rviz2-4] [INFO] [1713920514.696627029] [rviz2.moveit.ros.planning_scene_monitor]: Listening to '/moveit_cpp/publish_planning_scene'
[rviz2-4] [INFO] [1713920519.748710400] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
```
Generally this is what happens every time, but the "run_moveit_cpp" node dies rarely before the last row's fail.
I've checked a few issues regarding the "No kinematics plugins defined.", but with no improvement.
What could be a solution?

Thanks in advance!

sjahr, 2024-04-24:
MoveIt 2 had some breaking changes since Foxy. I'd recommend taking a look at the [migration guide](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) or compare your configs with more up-to-date configs in the [moveit_resources repository](https://github.com/ros-planning/moveit_resources/tree/humble) to make sure your config is correct.

sjahr, 2024-06-06:
Closing this because the issue seems to be solved

closed 2024-06-06 by sjahr