moveit/moveit2 #2815: Action server: /recognize_objects not available

LeonYo153, 2024-04-27:
### Description

When I launch the demo:
ros2 launch moveit2_tutorials demo.launch.py
the rviz appeared and immediately crush
with the output below:

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source
* If binary, which release version?
* If source, which branch?
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.

### Steps to reproduce
follow the tutorials to build the wrokspace

### Expected behaviour
see a world in RViz:

### Actual behaviour
the rviz appeared and immediately crush

### Backtrace or Console output

[rviz2-1] [WARN] [1714221707.900148700] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [spawner-8]: process has finished cleanly [pid 75722]
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-1] [INFO] [1714221708.048480698] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.00976645 seconds
[rviz2-1] [INFO] [1714221708.048611607] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1714221708.048642619] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [ERROR] [1714221711.646211568] [rviz2.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1714221711.692594016] [rviz2.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1714221711.801968741] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.00811746 seconds
[rviz2-1] [INFO] [1714221711.802047709] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1714221711.802060435] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1714221713.347932223] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'manipulator': 1 1 1 1 1 1 1
[rviz2-1] malloc(): invalid size (unsorted)
[ERROR] [rviz2-1]: process has died [pid 75708, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/yo/ws_moveit/install/moveit2_tutorials/share/moveit2_tutorials/launch/kinova_moveit_config_demo.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_yhgzlh04 --params-file /tmp/launch_params_qx59vxtl --params-file /tmp/launch_params_vxl13f2h --params-file /tmp/launch_params_p8ppoifw --params-file /tmp/launch_params_ftwc50pj'].
[image]

wc-richtech, 2024-05-11:
I have the same problem, ubuntu22.04, humble version.

```shell
ubuntu@ubuntu-Standard-PC-Q35-ICH9-2009:~/ws_moveit$ ros2 launch moveit2_tutorials demo.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-11-09-46-40-763357-ubuntu-Standard-PC-Q35-ICH9-2009-156950
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [156951]
[INFO] [static_transform_publisher-2]: process started with pid [156953]
[INFO] [robot_state_publisher-3]: process started with pid [156955]
[INFO] [move_group-4]: process started with pid [156975]
[INFO] [ros2_control_node-5]: process started with pid [156977]
[INFO] [spawner-6]: process started with pid [156979]
[INFO] [spawner-7]: process started with pid [156983]
[INFO] [spawner-8]: process started with pid [157002]
[static_transform_publisher-2] [INFO] [1715435201.051838275] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[robot_state_publisher-3] [INFO] [1715435201.054910897] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1715435201.054990405] [robot_state_publisher]: got segment bracelet_link
[robot_state_publisher-3] [INFO] [1715435201.055000388] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-3] [INFO] [1715435201.055004020] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1715435201.055007024] [robot_state_publisher]: got segment half_arm_1_link
[robot_state_publisher-3] [INFO] [1715435201.055009760] [robot_state_publisher]: got segment half_arm_2_link
[robot_state_publisher-3] [INFO] [1715435201.055012627] [robot_state_publisher]: got segment robotiq_85_base_link
[robot_state_publisher-3] [INFO] [1715435201.055015779] [robot_state_publisher]: got segment robotiq_85_left_finger_link
[robot_state_publisher-3] [INFO] [1715435201.055018607] [robot_state_publisher]: got segment robotiq_85_left_finger_tip_link
[robot_state_publisher-3] [INFO] [1715435201.055021909] [robot_state_publisher]: got segment robotiq_85_left_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1715435201.055025032] [robot_state_publisher]: got segment robotiq_85_left_knuckle_link
[robot_state_publisher-3] [INFO] [1715435201.055027867] [robot_state_publisher]: got segment robotiq_85_right_finger_link
[robot_state_publisher-3] [INFO] [1715435201.055031468] [robot_state_publisher]: got segment robotiq_85_right_finger_tip_link
[robot_state_publisher-3] [INFO] [1715435201.055034213] [robot_state_publisher]: got segment robotiq_85_right_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1715435201.055037304] [robot_state_publisher]: got segment robotiq_85_right_knuckle_link
[robot_state_publisher-3] [INFO] [1715435201.055040506] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1715435201.055043872] [robot_state_publisher]: got segment spherical_wrist_1_link
[robot_state_publisher-3] [INFO] [1715435201.055047065] [robot_state_publisher]: got segment spherical_wrist_2_link
[robot_state_publisher-3] [INFO] [1715435201.055049859] [robot_state_publisher]: got segment world
[ros2_control_node-5] [INFO] [1715435201.087527668] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-5] [INFO] [1715435201.088351481] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-5] [INFO] [1715435201.093244773] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-5] [INFO] [1715435201.107967081] [controller_manager]: Received robot description file.
[ros2_control_node-5] [INFO] [1715435201.108783284] [resource_manager]: Loading hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1715435201.109308521] [resource_manager]: Initialize hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [WARN] [1715435201.109332094] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [WARN] [1715435201.109359026] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example:
[ros2_control_node-5] <state_interface name="velocity">
[ros2_control_node-5]   <param name="initial_value">0.0</param>
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1715435201.109366891] [resource_manager]: Successful initialization of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1715435201.109384160] [resource_manager]: Loading hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1715435201.109393737] [resource_manager]: Initialize hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [WARN] [1715435201.109400751] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [INFO] [1715435201.109412709] [resource_manager]: Successful initialization of hardware 'RobotiqGripperHardwareInterface'
[... cut ...]
[rviz2-1] rviz2: malloc.c:4302: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.
[ERROR] [rviz2-1]: process has died [pid 156951, exit code -6, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/ubuntu/ws_moveit/install/moveit2_tutorials/share/moveit2_tutorials/launch/kinova_moveit_config_demo.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_1bvpscs1 --params-file /tmp/launch_params_4_54utj0 --params-file /tmp/launch_params_ig1s_f5q --params-file /tmp/launch_params_eercvu_u --params-file /tmp/launch_params_tjiqeoih'].

```

wc-richtech, 2024-05-11:
```shell
ubuntu@ubuntu-Standard-PC-Q35-ICH9-2009:~/ws_moveit$ /opt/ros/humble/lib/rviz2/rviz2 -d /home/ubuntu/ws_moveit/install/moveit2_tutorials/share/moveit2_tutorials/launch/kinova_moveit_config_demo.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_s_tr_6ui --params-file /tmp/launch_params_qj_zj12k --params-file /tmp/launch_params_jsmb5xa7 --params-file /tmp/launch_params_3muz7i0m --params-file /tmp/launch_params_zsv4jlms
[INFO] [1715435547.244263470] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1715435547.244428955] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1715435547.281778136] [rviz2]: Stereo is NOT SUPPORTED
[WARN] [1715435547.340889894] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
         at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[INFO] [1715435547.359902349] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.00192538 seconds
[INFO] [1715435547.359950110] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[INFO] [1715435547.359959649] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ERROR] [1715435550.562246892] [rviz2.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[INFO] [1715435550.567374320] [rviz2.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[INFO] [1715435550.612593267] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.00146892 seconds
[INFO] [1715435550.612633271] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[INFO] [1715435550.612640799] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[INFO] [1715435550.705596087] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'manipulator': 1 1 1 1 1 1 1
Segmentation fault (core dumped)

```
Run failed command, it shows

zp1915, 2024-06-17:
Ubuntu22.04, humble version.facing the same problem
zpvisen@zp-vm:~/ws_moveit2$ ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
[INFO] [launch]: All log files can be found below /home/zpvisen/.ros/log/2024-06-17-22-26-15-109065-zp-vm-48806
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [48807]
[INFO] [static_transform_publisher-2]: process started with pid [48809]
[INFO] [robot_state_publisher-3]: process started with pid [48811]
[INFO] [move_group-4]: process started with pid [48813]
[INFO] [ros2_control_node-5]: process started with pid [48815]
[INFO] [spawner-6]: process started with pid [48817]
[INFO] [spawner-7]: process started with pid [48819]
[INFO] [spawner-8]: process started with pid [48821]
[static_transform_publisher-2] [INFO] [1718634375.718782641] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[robot_state_publisher-3] [INFO] [1718634375.766960137] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1718634375.767126057] [robot_state_publisher]: got segment bracelet_link
[robot_state_publisher-3] [INFO] [1718634375.767145927] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-3] [INFO] [1718634375.767158114] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1718634375.767170220] [robot_state_publisher]: got segment half_arm_1_link
[robot_state_publisher-3] [INFO] [1718634375.767182037] [robot_state_publisher]: got segment half_arm_2_link
[robot_state_publisher-3] [INFO] [1718634375.767192632] [robot_state_publisher]: got segment robotiq_85_base_link
[robot_state_publisher-3] [INFO] [1718634375.767204618] [robot_state_publisher]: got segment robotiq_85_left_finger_link
[robot_state_publisher-3] [INFO] [1718634375.767216329] [robot_state_publisher]: got segment robotiq_85_left_finger_tip_link
[robot_state_publisher-3] [INFO] [1718634375.767227809] [robot_state_publisher]: got segment robotiq_85_left_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1718634375.767238963] [robot_state_publisher]: got segment robotiq_85_left_knuckle_link
[robot_state_publisher-3] [INFO] [1718634375.767248648] [robot_state_publisher]: got segment robotiq_85_right_finger_link
[robot_state_publisher-3] [INFO] [1718634375.767261143] [robot_state_publisher]: got segment robotiq_85_right_finger_tip_link
[robot_state_publisher-3] [INFO] [1718634375.767273085] [robot_state_publisher]: got segment robotiq_85_right_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1718634375.767285099] [robot_state_publisher]: got segment robotiq_85_right_knuckle_link
[robot_state_publisher-3] [INFO] [1718634375.767295150] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1718634375.767307005] [robot_state_publisher]: got segment spherical_wrist_1_link
[robot_state_publisher-3] [INFO] [1718634375.767318909] [robot_state_publisher]: got segment spherical_wrist_2_link
[robot_state_publisher-3] [INFO] [1718634375.767330272] [robot_state_publisher]: got segment world
[move_group-4] [INFO] [1718634375.780841098] [move_group.moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.0114496 seconds
[move_group-4] [INFO] [1718634375.781016668] [move_group.moveit.moveit.core.robot_model]: Loading robot model 'gen3'...
[move_group-4] [INFO] [1718634375.781037417] [move_group.moveit.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ros2_control_node-5] [INFO] [1718634375.783697892] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-5] [INFO] [1718634375.785721648] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-5] [INFO] [1718634375.786133704] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-5] [INFO] [1718634375.797202258] [controller_manager]: Received robot description file.
[ros2_control_node-5] [INFO] [1718634375.797973851] [resource_manager]: Loading hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1718634375.799097638] [resource_manager]: Initialize hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [WARN] [1718634375.799185936] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [WARN] [1718634375.799232600] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1718634375.799252521] [resource_manager]: Successful initialization of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1718634375.799286525] [resource_manager]: Loading hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1718634375.799306056] [resource_manager]: Initialize hardware 'RobotiqGripperHardwareInterface' 
[... cut ...]
[rviz2-1] [INFO] [1718634380.361503360] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'manipulator': 1 1 1 1 1 1 1
[ERROR] [rviz2-1]: process has died [pid 48807, exit code -7, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d /home/zpvisen/ws_moveit2/install/moveit2_tutorials/share/moveit2_tutorials/launch/kinova_moveit_config_demo.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_ducupnut --params-file /tmp/launch_params_xel2kged --params-file /tmp/launch_params_p1jxa0bf --params-file /tmp/launch_params_3uh81rmf --params-file /tmp/launch_params_11c0fi4y'].

Markus-Simonsen, 2024-09-20:
I am getting the same error message with: 
`Action server: /recognize_objects not available`
However for me it does not crash

zhuxiyu24, 2024-11-07:
俺也一样

mikeferguson, 2024-12-17:
This message is a red herring - it prints for basically everyone because nobody out there is actually using the recognize_objects feature - but it isn't the source of your crash.

Most of the crashes above are segmentation faults - and these are almost certainly due to mixing outdated Debians with current source branch. We recently released updated Humble Debians because an ABI change had been made - please update your Debian install, or better yet, remove it entirely when building from source.

DrMax1, 2024-12-19:
> This message is a red herring - it prints for basically everyone because nobody out there is actually using the recognize_objects feature - but it isn't the source of your crash.
>
> Most of the [...]

After completly removing and reinstalling the latest version (binary) for humble, this exact issue still persists. Are there any other solutions? This makes MoveIt unusable for me at the moment

mikeferguson, 2024-12-19:
@DrMax1 instead of jumping on a closed ticket, please start a new one with your exact details/backtrace of the crash