moveit/moveit2 #2821: Re-enable `planning_scene_monitor.test.py` test after diagnosing its flaky bring-up

EzraBrooks, 2024-04-30:
This is failing across a number of branches. It may be a latency issue, esp. if that latency is induced specifically by the slowness of GitHub Actions runners, since we can't seem to reproduce this locally.

sea-bass, 2024-11-17:
It appears this test is still enabled on the `humble` branch and very much flaky.

Rather than disabling it there, I wonder if it's a good time to give this a look.

sea-bass, 2024-11-23:
Note that I've re-enabled the test after adding a 1 second sleep in the fixture setup, which somehow resolves a race condition and seems to make the test pass consistently (I ran them many times in CI).

https://github.com/moveit/moveit2/pull/3124

I would not consider this issue closed until the real root cause is resolved, though.