moveit/moveit2 #2824: Bullet collision checking not working

MatthijsBurgh, 2024-05-03:
### Description
When I configure moveit to use Bullet collision detection, it already thinks my start state is in collision, while the same robot state with FCL doesn't think it is collision

```yaml
# moveit_cpp.yaml
collision_detector: Bullet
```
I am using `moveit_py`.  Besides the log mentioned below., I don't get any other logs. I have event tried a local build of `moveit_py`, so I could add the `--log-level moveit.core.collision_detection_bullet:=debug` args. But it doesn't provided me with any informative logs.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Binary build: 2.9.0-1
* Which RMW: Cyclone

### Steps to reproduce
- Put your robot in a non collision state
- Try to plan with the `default_planning_request_adapters/CheckStartStateCollision` enabled.

### Expected behaviour
Valid collision checking

### Actual behaviour
Thinks it is in collision, while it is not.

### Backtrace or Console output
```py
PlanningRequestAdapter 'CheckStartStateCollision' failed, because 'Start state in collision.'. Aborting planning pipeline.
```

sjahr, 2024-06-13:
@MatthijsBurgh Thanks for reporting this. Do you have a setup available that could be used to reproduce the issue?

MatthijsBurgh, 2024-06-20:
@sjahr sorry for the late response. I don't any easy setup for to test, sorry. I am using a TechMan robot, the robot description and moveit config can be found [here](https://github.com/MatthijsBurgh/tmr_ros2). I am using the TM14 robot.You should change the `moveit_cpp.yaml` config to use bullet. Like the code snippet at the start of this issue.