moveit/moveit2 #2837: MoveIt2 Setup Assistant generated demo.launch.py error

nilp-dromeda, 2024-05-17:
### Description

After correctly following the MoveIt2 setup assistant [tutorial](https://moveit.picknik.ai/main/doc/examples/setup_assistant/setup_assistant_tutorial.html), I decided to checkout if the output of the setup assistant was properly working. When running `demo.launch.py` it always ends in an error.

### Your environment
* ROS Distro: [Humble
* OS Version: Ubuntu 20.04
* Using NixOS to build
* Source from this [NixOS repo](https://github.com/lopsided98/nix-ros-overlay)

### Steps to reproduce
1. Use Moveit Setup Assistant to generate moveit_config package.
2. Run `demo.launch.py`

### Expected behaviour
Rviz should pop up and nothing should crash.

### Actual behaviour
Do not even get to the point of launching Rviz. Program is exited with the below error:
```bash
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Expected 'value' to be one of [<class 'float'>, <class 'int'>, <class 'str'>, <class 'bool'>, <class 'bytes'>], but got '()' of type '<class 'tuple'>'
```

itohemon, 2024-06-29:
I also ran into the same message when I went through the [setup_assistant_tutorial](https://moveit.picknik.ai/main/doc/examples/setup_assistant/setup_assistant_tutorial.html).

I fixed the message by modifying config/moveit_controllers.yaml as follows:
```yaml
# MoveIt uses this configuration for controller management

moveit_controller_manager: moveit_simple_controller_manager/MoveItSimpleControllerManager

moveit_simple_controller_manager:
  controller_names:
    - panda_arm_controller
    - hand_controller

  panda_arm_controller:
    type: FollowJointTrajectory
    action_ns: follow_joint_trajectory
    default: true
    joints:
      - panda_joint1
      - panda_joint2
      - panda_joint3
      - panda_joint4
      - panda_joint5
      - panda_joint6
      - panda_joint7
    action_ns: follow_joint_trajectory
    default: true
  hand_controller:
    type: GripperCommand
#    joints:  # Comment out this line
#      []     # Comment out this line
    action_ns: gripper_cmd
    default: true
```

It seems that it is not good that the joint of the hand_controller in the moveit_controllers is empty [here](https://moveit.picknik.ai/main/doc/examples/setup_assistant/setup_assistant_tutorial.html#step-10-moveit-controllers
).

zhaoyachuanl, 2024-09-18:
> I also ran into the same message when I went through the [setup_assistant_tutorial](https://moveit.picknik.ai/main/doc/examples/setup_assistant/setup_assistant_tutorial.html).
>
> I fixed the message [...]

thanks,it s really useful

kevin00156, 2024-11-14:
I have same issue, too. And it's useful.

Environment
ROS Distro: Humble
OS Version: Ubuntu 22.04
Using WSL2 in Windows 11