moveit/moveit2 #2841: GSOC 2024: MuJoCo ROS/MoveIt integration

sangteak601, 2024-05-18:
Hi all,

I will work on integrating MuJoCo simulation with ROS2 and MoveIt2 for GSOC. The project description can be found [here](https://summerofcode.withgoogle.com/programs/2024/projects/tVGym4TK)

The goals and milestones of the project will be as follows:
- [ ] Implement ROS2 interface for MuJoCo(`mujoco_ros2_control`)
- [ ] Create example use-case using `mujoco_ros2_control` and MoveIt
- [ ] Compare the result with other simulation tools

peterdavidfagan, 2024-05-31:
Hi @sangteak601, 

It is great that you are participating in GSoC and working on this project. I was a past participant of GSoC with MoveIt, in general its a great experience, I hope you have a great time. 

I wished to post here as I have had the need to work on the topic of integrating mujoco with ROS as part of my workflow for performing robotics research. To this end I create the following package: https://github.com/peterdavidfagan/mujoco_ros. One aspect of this package that I find very useful is the integration of mujoco's interactive viewer. 

I am very interested in the work you will be completing and would also be keen to align my workflows/packages with your ros2_control integration (right now I leverage topic_based_ros2_control). Looking forward to seeing the progress you make and if you ever want to chat about any of the details of what I have done to date on this topic I would be happy to discuss.

gavanderhoorn, 2024-05-31:
It's ROS 1, AFAIK, but @rhaschke's team/colleagues have been working on: [ubi-agni/mujoco_ros_pkgs](https://github.com/ubi-agni/mujoco_ros_pkgs).

peterdavidfagan, 2024-05-31:
+1 on @gavanderhoorn @rhaschke's teams work is really nice on this. Also maybe it came up already in meetings as @sjahr mentioned this to me in the past but there also the following: https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/tree/ros2/cartesian_controller_simulation.

sangteak601, 2024-06-23:
Hi @peterdavidfagan and @gavanderhoorn,
Thanks for your interest in my GSOC project and for sharing other relevant repos. It’s good to know there are some workarounds on this topic. They all look very useful! My work will be updated on my personal repo (https://github.com/sangteak601/mujoco_ros2_control). If you’re interested, please check it out for progress.