moveit/moveit2 #2842: stderr: moveit_core

ostifede02, 2024-05-19:
Hello!
Unfortunately I get a similar issue. I have re-installed ROS2 and followed the tutorial from [here](https://moveit.ros.org/install-moveit2/source/).

Thank you in advance for your help.

`--- stderr: moveit_core
CMake Error at /opt/ros/humble/share/kdl_parser/cmake/export_kdl_parserExport.cmake:56 (set_target_properties):
  The link interface of target "kdl_parser::kdl_parser" contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/kdl_parser/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/kdl_parser/cmake/kdl_parserConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)

CMake Error at /opt/ros/humble/share/srdfdom/cmake/srdfdomTargetsExport.cmake:56 (set_target_properties):
  The link interface of target "srdfdom::srdfdom" contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Error at /opt/ros/humble/share/urdf/cmake/urdfExport.cmake:56 (set_target_properties):
  The link interface of target "urdf::urdf" contains:

    urdfdom::urdfdom_sensor

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/urdf/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/urdf/cmake/urdfConfig.cmake:41 (include)
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Error at /opt/ros/humble/share/urdf_parser_plugin/cmake/urdf_parser_plugin-exportExport.cmake:56 (set_target_properties):
  The link interface of target "urdf_parser_plugin::urdf_parser_plugin"
  contains:

    urdfdom_headers::urdfdom_headers

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/humble/share/urdf_parser_plugin/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /opt/ros/humble/share/urdf_parser_plugin/cmake/urdf_parser_pluginConfig.cmake:41 (include)
  /opt/ros/humble/share/urdf/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/urdf/cmake/urdfConfig.cmake:41 (include)
  /opt/ros/humble/share/srdfdom/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/srdfdom/cmake/srdfdomConfig.cmake:41 (include)
  CMakeLists.txt:29 (find_package)

CMake Generate step failed.  Build files cannot be regenerated correctly.
---
Failed   <<< moveit_core [2.31s, exited with code 1]

`

sjahr, 2024-06-13:
@ostifede02 Are you still experiencing the issue? On which operating system are you compiling and which branch / commit are you using?

sjahr, 2024-07-30:
Closing due to inactivity. Feel free to re-open if you run into it again

closed 2024-07-30 by sjahr