moveit/moveit2 #2843: MoveIt Servo includes a default `joint_limit_margins` that only works with a 7dof arm

EzraBrooks, 2024-05-22:
### Description

The default value for `joint_limit_margins` in Servo is `[0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]`.

https://github.com/moveit/moveit2/blob/c72ac1bfc4fad1288bc0e5361c16edaaf8b90a01/moveit_ros/moveit_servo/config/servo_parameters.yaml#L327-L334

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 20.04
* Source @ f02aa93
* CycloneDDS

### Steps to reproduce

- Attempt to run Servo against an arm of any dofs other than 7 without overriding `joint_limit_margins`.

### Expected behaviour

Either:

The default value should be auto-populated as a vector of the same length as the active move group that contains `0.1` in each index,

or 

No default value should be set, and the user should be informed that they must set `joint_limit_margins`.

### Actual behaviour

Servo crashes with an error message telling you that you must have the same number of elements in `joint_limit_margins` and your active move group. Since you didn't set `joint_limit_margins`, it isn't present in the "parameters YAML file" mentioned in the error message, so this is a bit confusing.

### Backtrace or Console output

(sorry, would include as text, but was sent as a screenshot to me)

[image]

scastro-bdai, 2024-05-23:
Yes! I think the default should be a scalar value that expands to the length of the actual number of DOFs if that is detected.

dohyeoklee, 2024-06-25:
Are you still suffering from this issue?
For me, I also got the same kind of error, so I changed the source code and re-build it.
(I use ur5 so I changed [0.1,0.1,0.1,0.1,0.1,0.1,0.1] to [0.1,0.1,0.1,0.1,0.1,0.1])
It actually solved the problem but another error came out.
So I re-installed everything from ubuntu to moveit.
(original: ubuntu 22.04, ros2 iron, moveit2 from source build
-> alternative: ubuntu 22.04, ros2 humble, ros2 middleware as cyclone dds, moveit2 from binary install, ros2 control install)
It still causes joint_margin error, but everything works perfect. (no crash)
From my guess, the joint_margin error itself is not the cause of this issue, it's just the result.
So I recommend you to re-install and change some install methods.

closed 2024-08-15 by sjahr