moveit/moveit2 #2846: Jazzy Support and Release

henningkayser, 2024-05-23:
[ROS 2 Jazzy is being released these days](https://docs.ros.org/en/jazzy/Releases/Release-Jazzy-Jalisco.html#release-timeline), and we are working on adding support for it in MoveIt, pushing for having the Source (and soon Debian) release version completed. The plan is to enable Noble CI jobs in all repositories, testing Jazzy along with Rolling, and applying the necessary remaining fixes. Thanks to @rhaschke a first step to fixing Noble support has been already provided with https://github.com/moveit/moveit2/pull/2793. The remaining steps for completion are:

- [ ] Fix MoveIt docker images
- [ ] Add Jazzy build job to MoveIt and dependencies
- [ ] Update README and build instructions
- [ ] Cut release for Rolling and Jazzy

rhaschke, 2024-05-23:
What fixes do you have in mind for the MoveIt docker images, @henningkayser?
With #2793, I thought I have fixed all open issues. I just noticed that - due to the ros-planning org migration, dockerhub doesn't yet receive image updates. That's tackled in #2847.

henningkayser, 2024-06-14:
Jazzy: https://github.com/ros/rosdistro/pull/41645
Rolling: https://github.com/ros/rosdistro/pull/41643

zouzhe1, 2024-06-24:
How long can we release Moveit2 that supports Noble versions? thanks   @henningkayser

henningkayser, 2024-07-02:
Closing this, the broken tutorial issue is being tracked here https://github.com/moveit/moveit2/issues/2864

closed 2024-07-02 by henningkayser

usmanNoor5, 2024-08-28:
can i  install ubuntu 24.04 on jetson nano with ros2 jazzy with all that can i use isaac packages with it