moveit/moveit2 #2848: GSOC 2024: Moveit Drake integration experiment

kamiradi, 2024-05-23:
Hi,

This issue lists and tracks the moveit-drake experimental integration project. I propose the following ideas, 

- [ ] Exposing the `KinematicTrajectoryOptimization` [functionality](https://drake.mit.edu/doxygen_cxx/classdrake_1_1planning_1_1trajectory__optimization_1_1_kinematic_trajectory_optimization.html#aa04be32847a0a206b2ae988625279139) as a moveit motion planning plugin.
- [ ] Exposing `GlobalInverseKinematics` as an IK plugin.
- [ ] Exposing the `DoDifferentialInverseKinematics()` [functionality](https://drake.mit.edu/doxygen_cxx/structdrake_1_1multibody_1_1_differential_inverse_kinematics_result.html) into the moveit ecosystem. 
- [ ] Necessary tutorials and documentation on writing your own drake based moveit motion planner.
Since this is an experimental project, I am open to other ideas that could serve as a benchmark for motion planning in moveit using drake.

rhaschke, 2024-05-23:
To apply Drake's motion planning framework, the first step will be to translate MoveIt's PlanningScene to Drake's representation. Do you already have a roadmap for this?

kamiradi, 2024-06-10:
Thanks for your interest @rhaschke,

Indeed, this is the crux of the experimentation as you have rightly foreseen. Currently, I am focussing on transcribing a simple motion planning problem from Moveit to Drake and reporting the result. While Moveit's representation of the robot employs the RobotModel class, Drake uses the MultibodyPlant. Similarly, Moveit's PlanningScene finds its ideological analogue in Drake's SceneGraph representation. Either way, there is the additional transcription of utilising the SceneGraph to infer constraints for the Trajectory Optimization instance. I plan to build V1 of this integration with the following steps

1.  Create mirrored objects of RobotModel and PlanningScene by instantiating a MultibodyPlant and SceneGraph class within the Moveit plugin. 
2. Utilise the drake classes to setup and solve the optimization problem.
3. Report the result using Moveit's DisplayTrajectory.

I am open to ideas you may have otherwise that makes setting up the optimization problem easier, and if I am missing something obvious.

Looking forward to the discussion.

sea-bass, 2025-02-02:
I think we can call this complete with the creation of https://github.com/moveit/moveit_drake 🎉

closed 2025-02-02 by sea-bass