moveit/moveit2 #2853: py_binding_tools

robinshie, 2024-05-25:
Environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04

### Description
I got error as folow:

Finished <<< moveit_ros [0.87s]                                              
--- stderr: moveit_task_constructor_core                                     
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findpy_binding_tools.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "py_binding_tools", but CMake did not find one.

  Could not find a package configuration file provided by "py_binding_tools"
  with any of the following names:

    py_binding_toolsConfig.cmake
    py_binding_tools-config.cmake

  Add the installation prefix of "py_binding_tools" to CMAKE_PREFIX_PATH or
  set "py_binding_tools_DIR" to a directory containing one of the above
  files.  If "py_binding_tools" provides a separate development package or
  SDK, be sure it has been installed.

---

rhaschke, 2024-05-26:
Fixed in MTC.

closed 2024-05-26 by rhaschke

OdedHorowits, 2024-07-23:
>Fixed in MTC.

Hi
What does in mean?
I just pulled from `humble` and still get this error

GraemeJf, 2024-07-24:
I am also having this same issue with both the Humble and ROS2 branch.

rhaschke, 2024-07-25:
Either install py_binding_tools from binary packages (ros-<rosdistro>-py-binding-tools) or from source: https://github.com/moveit/py_binding_tools

aymanT122R, 2025-03-06:
@rhaschke this packages provided for ros1 where I found the ros2 branch

rhaschke, 2025-03-06:
> this packages provided for ros1 where I found the ros2 branch

@aymanT122R: I don't understand this comment or question. py_binding_tools is available (and released) for both, ROS1 and ROS2.