moveit/moveit2 #2854: 'joint_state_broadcaster' [move_group-4] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'

xuanmumy, 2024-05-26:
Hello, 
I follow "MoveIt Quickstart in RViz" tutorial to run ros2 launch moveit2_tutorials demo.launch.py  after build moveit.
But "displays panel" shows warning "Requesting initial scene failed" both under PlanningScene and MotionPlanning dir. 

### Steps to reproduce
follow "MoveIt Quickstart in RViz[](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#moveit-quickstart-in-rviz)"
until ros2 launch moveit2_tutorials demo.launch.py
### Expected behaviour
display is not same as tutorial.

### Actual behaviour
warnning displayed 

### Backtrace or Console output
[ros2_control_node-5] [INFO] [1716735893.396328900] [controller_manager]: Loading controller 'joint_state_broadcaster'
[move_group-4] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[move_group-4]   what():  expected [string_array] got [string]
[move_group-4] Stack trace (most recent call last):
[move_group-4] #17   Object "", at 0xffffffffffffffff, in
[move_group-4] #16   Object "/home/sfl/workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x7fcceb185094, in _start
[move_group-4] #15   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fccea699e3f, in __libc_start_main
[move_group-4] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fccea699d8f, in
[move_group-4] #13   Object "/home/sfl/workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x7fcceb18421e, in main
[move_group-4] #12   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7fcceb00f287, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-4] #11   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7fcceb00e28b, in moveit_cpp::MoveItCpp::loadPlanningPipelines(moveit_cpp::MoveItCpp::PlanningPipelineOptions const&)
[move_group-4] #10   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline_interfaces.so.2.9.0", at 0x7fccea3e6c76, in moveit::planning_pipeline_interfaces::createPlanningPipelineMap(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #9    Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.9.0", at 0x7fccea261d96, in planning_pipeline::PlanningPipeline::PlanningPipeline(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #8    Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.9.0", at 0x7fccea2a5ff3, in planning_pipeline_parameters::ParamListener::declare_params()
[move_group-4] #7    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fcceac01863, in
[move_group-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fccea96e4d7, in __cxa_throw
[move_group-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fccea96e276, in std::terminate()
[move_group-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fccea96e20b, in
[move_group-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fccea962b9d, in
[move_group-4] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fccea6987f2, in abort
[move_group-4] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fccea6b2475, in raise
[move_group-4] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fccea7069fc, in pthread_kill
[move_group-4] Aborted (Signal sent by tkill() 9615 1000)
....
[ros2_control_node-5] [INFO] [1716735893.418056000] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-6] [INFO] [1716735893.431671600] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[rviz2-1] [WARN] [1716735893.447670900] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
...
[rviz2-1] [INFO] [1716735900.019674700] [rviz2.moveit.ros.robot_interaction]: No active joints or end effectors found for group 'gripper'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-1] [INFO] [1716735900.020273600] [rviz2.moveit.ros.robot_interaction]: No active joints or end effectors found for group 'gripper'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-1] [INFO] [1716735904.898212400] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1716735905.076620500] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

Could you help me to find out what is the problem?
thank you, I am a freshman on this.

xuanmumy, 2024-05-26:
Sorry, I missed an error log: there is an error before " terminate called...." line
------------------------------------------------------------------------------------------------
[move_group-4] [INFO] [1716737911.250374500] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-4] [WARN] [1716737911.281109100] [move_group.moveit.moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-4] [ERROR] [1716737911.281135600] [move_group.moveit.moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[move_group-4] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[move_group-4]   what():  expected [string_array] got [string]
[move_group-4] Stack trace (most recent call last):
[move_group-4] #17   Object "", at 0xffffffffffffffff, in
[move_group-4] #16   Object "/home/sfl/workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x7fdbc32ca094, in _start
[move_group-4] #15   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbc27e9e3f, in __libc_start_main
[move_group-4] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbc27e9d8f, in
[move_group-4] #13   Object "/home/sfl/workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x7fdbc32c921e, in main
[move_group-4] #12   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7fdbc315f287, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-4] #11   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7fdbc315e28b, in moveit_cpp::MoveItCpp::loadPlanningPipelines(moveit_cpp::MoveItCpp::PlanningPipelineOptions const&)
[move_group-4] #10   Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline_interfaces.so.2.9.0", at 0x7fdbc2536c76, in moveit::planning_pipeline_interfaces::createPlanningPipelineMap(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #9    Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.9.0", at 0x7fdbc23b1d96, in planning_pipeline::PlanningPipeline::PlanningPipeline(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #8    Object "/home/sfl/workspace/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.9.0", at 0x7fdbc23f5ff3, in planning_pipeline_parameters::ParamListener::declare_params()
[move_group-4] #7    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7fdbc2d51863, in
[move_group-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fdbc2abe4d7, in __cxa_throw
[move_group-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fdbc2abe276, in std::terminate()
[move_group-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fdbc2abe20b, in
[move_group-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fdbc2ab2b9d, in
[move_group-4] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbc27e87f2, in abort
[move_group-4] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbc2802475, in raise
[move_group-4] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fdbc28569fc, in pthread_kill
[move_group-4] Aborted (Signal sent by tkill() 9854 1000)
[ERROR] [move_group-4]: process has died [pid 9854, exit code -6, cmd '/home/sfl/workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_0lrono0x'].
[rviz2-1] [INFO] [1716737911.397142500] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1716737911.397297400] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-1] [INFO] [1716737911.505383600] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [INFO] [1716737911.531600800] [controller_manager]: Loading controller 'joint_trajectory_controller'

xuanmumy, 2024-05-26:
[rviz2-1] [INFO] [1716737911.505383600] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-5] [INFO] [1716737911.531600800] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ros2_control_node-5] [WARN] [1716737911.564460900] [joint_trajectory_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[ros2_control_node-5] [INFO] [1716737911.566139300] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-7] [INFO] [1716737911.566845100] [spawner_joint_trajectory_controller]: Loaded joint_trajectory_controller
[ros2_control_node-5] [INFO] [1716737911.581092700] [controller_manager]: Configuring controller 'joint_trajectory_controller'
[ros2_control_node-5] [INFO] [1716737911.581246700] [joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-5] [INFO] [1716737911.581263400] [joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-5] [INFO] [1716737911.581271700] [joint_trajectory_controller]: Using 'splines' interpolation method.
[spawner-6] [INFO] [1716737911.581710800] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-5] [INFO] [1716737911.585815800] [joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ros2_control_node-5] [INFO] [1716737911.599737400] [joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2_control_node-5] [INFO] [1716737911.611640700] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-5] [INFO] [1716737911.611698400] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-7] [INFO] [1716737911.616625900] [spawner_joint_trajectory_controller]: Configured and activated joint_trajectory_controller
[spawner-6] [INFO] [1716737911.618711000] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-7]: process has finished cleanly [pid 9860]
[INFO] [spawner-6]: process has finished cleanly [pid 9858]
[rviz2-1] [WARN] [1716737912.054396200] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-1]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[rviz2-1] [INFO] [1716737912.166645900] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.0105669 seconds
[rviz2-1] [INFO] [1716737912.166696200] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1716737912.166712400] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[spawner-8] [INFO] [1716737913.269129600] [spawner_robotiq_gripper_controller]: Waiting for '/controller_manager' node to exist
[ros2_control_node-5] [INFO] [1716737914.110154300] [controller_manager]: Loading controller 'robotiq_gripper_controller'
[spawner-8] [INFO] [1716737914.300750700] [spawner_robotiq_gripper_controller]: Loaded robotiq_gripper_controller
[ros2_control_node-5] [INFO] [1716737914.305989400] [controller_manager]: Configuring controller 'robotiq_gripper_controller'
[ros2_control_node-5] [INFO] [1716737914.306063400] [robotiq_gripper_controller]: Action status changes will be monitored at 20Hz.
[spawner-8] [INFO] [1716737914.344855500] [spawner_robotiq_gripper_controller]: Configured and activated robotiq_gripper_controller
[INFO] [spawner-8]: process has finished cleanly [pid 9862]
[rviz2-1] [ERROR] [1716737915.482470900] [rviz2.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1716737915.787330000] [rviz2.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1716737916.163093000] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.008961 seconds
[rviz2-1] [INFO] [1716737916.163143300] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1716737916.163154800] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1716737916.759905300] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'manipulator': 1 1 1 1 1 1 1
[rviz2-1] [INFO] [1716737918.281331500] [rviz2.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[rviz2-1] [INFO] [1716737918.298868700] [rviz2.moveit.ros.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[... cut ...]
[rviz2-1] [INFO] [1716737983.693887400] [rviz2.moveit.ros.move_group_interface]: Ready to take commands for planning group gripper.
[rviz2-1] [ERROR] [1716738025.718707500] [rviz2.moveit.ros.background_processing]: Exception caught while processing action 'Frame::changePlanningGroup': std::future_error: Broken promise
[rviz2-1] [INFO] [1716738025.723654800] [rviz2.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
[rviz2-1] [INFO] [1716738025.728048200] [rviz2.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor

Kaiser1401, 2024-06-04:
Found something for ros-jazzy:

1) The 3D sensor error is unrelated and can be ignored. it just happens to be directly before the real problems start and is therefore quite missleading

2) moveit (not sure which part exactly) in jazzy expects the "request_adapters" parameter to be a string_array again instead of a string. somehting that seems to have been changed the other way around in ros-humble. Causing "what(): expected [string_array] got [string]"

-> fix: in your moveit2_tutorial_ws in `ros2_kortex/kortex_moveit_config/kinova_gen3_7dof_robotiq_2f_85_moveit_config/config/ompl_planning.yaml` change 

```
request_adapters: >-
  default_planner_request_adapters/AddTimeOptimalParameterization
  default_planner_request_adapters/ResolveConstraintFrames
  default_planner_request_adapters/FixWorkspaceBounds
  default_planner_request_adapters/FixStartStateBounds
  default_planner_request_adapters/FixStartStateCollision
  default_planner_request_adapters/FixStartStatePathConstraints
```
to
```
request_adapters:
  - default_planner_request_adapters/AddTimeOptimalParameterization
  - default_planner_request_adapters/ResolveConstraintFrames
  - default_planner_request_adapters/FixWorkspaceBounds
  - default_planner_request_adapters/FixStartStateBounds
  - default_planner_request_adapters/FixStartStateCollision
  - default_planner_request_adapters/FixStartStatePathConstraints
```

this removes the error.

3) Next error is '[move_group-4]   what():  Planning plugin name is empty or not defined in namespace 'ompl'. Please choose one of the available plugins: chomp_interface/CHOMPPlanner, ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner, planning_pipeline_test/DummyPlannerManager, stomp_moveit/StompPlanner'

I am not sure why yet, cause the aforementioned yaml states `planning_plugin: ompl_interface/OMPLPlanner` which is also present in the parameters passed to move_group in the launchfile

Kaiser1401, 2024-06-06:
Fix for ROS Jazzy:

in the `moveit2_tutorial` workspace in `ros2_kortex/kortex_moveit_config/kinova_gen3_7dof_robotiq_2f_85_moveit_config/config/ompl_planning.yaml` ([original file](https://github.com/Kinovarobotics/ros2_kortex/blob/main/kortex_moveit_config/kinova_gen3_7dof_robotiq_2f_85_moveit_config/config/ompl_planning.yaml))

replace 

``` diff
- planning_plugin: ompl_interface/OMPLPlanner
- request_adapters: >-
-   default_planner_request_adapters/AddTimeOptimalParameterization
-   default_planner_request_adapters/ResolveConstraintFrames
-   default_planner_request_adapters/FixWorkspaceBounds
-   default_planner_request_adapters/FixStartStateBounds
-   default_planner_request_adapters/FixStartStateCollision
-   default_planner_request_adapters/FixStartStatePathConstraints
+ planning_plugins:
+   - ompl_interface/OMPLPlanner
+ # The order of the elements in the adapter corresponds to the order they are processed by the motion planning pipeline.
+ request_adapters:
+   - default_planning_request_adapters/ResolveConstraintFrames
+   - default_planning_request_adapters/ValidateWorkspaceBounds
+   - default_planning_request_adapters/CheckStartStateBounds
+   - default_planning_request_adapters/CheckStartStateCollision
+ response_adapters:
+   - default_planning_response_adapters/AddTimeOptimalParameterization
+   - default_planning_response_adapters/ValidateSolution
+   - default_planning_response_adapters/DisplayMotionPath
```

xuanmumy, 2024-06-08:
Many thanks, it looks works, no warning in panel now.

closed 2024-06-08 by xuanmumy

JenniferBuehler, 2024-06-27:
See also #2887