moveit/moveit2 #2887: move_group Process Dying in MoveIt2 Tutorials

Leopold-Klotz, 2024-06-27:
Hello all,

I was having an issue with move_group crashing when I was trying to launch a personal package which depended on moveit2. I tried to figure it out, but eventually could not find the root of the issue. Initially I thought that it must be something with my machine, but with a fresh install I'm now having issues with the provided moveit2 tutorials as well.

To try and find the issue, I have just freshly installed ubuntu 22.04 on my machine, and a new installation of both ROS2 Humble following the main instructions, as well as installing MoveIt2. When I try and go through the tutorial, I am getting an error where the move_group is process dying. 

Any advice would be appreciated and if I figure it out I will post the solution/fix the code.
From what I can tell, the relevant logs are:

[move_group-4] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[move_group-4]   what():  expected [string_array] got [string]
[move_group-4] Stack trace (most recent call last):
[move_group-4] #17   Object "", at 0xffffffffffffffff, in 
[move_group-4] #16   Object "/home/leo/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x5b58c1313094, in _start
[move_group-4] #15   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7efb48629e3f]
[move_group-4] #14   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7efb48629d8f]
[move_group-4] #13   Object "/home/leo/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x5b58c131221e, in main
[move_group-4] #12   Object "/home/leo/ws_moveit/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.10.0", at 0x7efb492b2287, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-4] #11   Object "/home/leo/ws_moveit/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.10.0", at 0x7efb492b128b, in moveit_cpp::MoveItCpp::loadPlanningPipelines(moveit_cpp::MoveItCpp::PlanningPipelineOptions const&)
[move_group-4] #10   Object "/home/leo/ws_moveit/install/moveit_ros_planning/lib/libmoveit_planning_pipeline_interfaces.so.2.10.0", at 0x7efb488fac76, in moveit::planning_pipeline_interfaces::createPlanningPipelineMap(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #9    Object "/home/leo/ws_moveit/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.10.0", at 0x7efb4886cd26, in planning_pipeline::PlanningPipeline::PlanningPipeline(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-4] #8    Object "/home/leo/ws_moveit/install/moveit_ros_planning/lib/libmoveit_planning_pipeline.so.2.10.0", at 0x7efb488b10b3, in planning_pipeline_parameters::ParamListener::declare_params()
[move_group-4] #7    Object "/opt/ros/humble/lib/librclcpp.so", at 0x7efb48eb2849, in 
[move_group-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7efb48aae4d7, in __cxa_throw
[move_group-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7efb48aae276, in std::terminate()
[move_group-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7efb48aae20b, in 
[move_group-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7efb48aa2b9d, in 
[move_group-4] #2    Source "./stdlib/abort.c", line 79, in abort [0x7efb486287f2]
[move_group-4] #1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7efb48642475]
[move_group-4] #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
[move_group-4]     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
[move_group-4]       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7efb486969fc]
[move_group-4] Aborted (Signal sent by tkill() 174093 1000)
[ERROR] [move_group-4]: process has died [pid 174093, exit code -6, cmd '/home/leo/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_yoqu6zsx'].

JenniferBuehler, 2024-06-27:
I confirm that I am experiencing the same. It's the same issue as [here](https://github.com/moveit/moveit2/issues/2854#issuecomment-2152224079) and this solution solved it for me.

manadaniel, 2024-06-27:
I am experiencing the same issue with the tutorials and other projects using the following setup:

- Ubuntu 24.04
- ROS2 Jazzy
- Moveit2 Main on commit 94796046b5ae6fc2a8e5f5ee73db4944d71a5643

Here is the terminal output:

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [5760]
[INFO] [static_transform_publisher-2]: process started with pid [5761]
[INFO] [robot_state_publisher-3]: process started with pid [5762]
[INFO] [move_group-4]: process started with pid [5763]
[INFO] [ros2_control_node-5]: process started with pid [5764]
[INFO] [spawner-6]: process started with pid [5765]
[INFO] [spawner-7]: process started with pid [5766]
[INFO] [spawner-8]: process started with pid [5767]
[static_transform_publisher-2] [INFO] [1719469827.767164578] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[robot_state_publisher-3] [INFO] [1719469827.774529923] [robot_state_publisher]: Robot initialized
[ros2_control_node-5] [INFO] [1719469827.784998003] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-5] [INFO] [1719469827.785051080] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-5] [WARN] [1719469827.785092366] [controller_manager]: Could not enable FIFO RT scheduling policy. Consider setting up your user to do FIFO RT scheduling. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details.
[ros2_control_node-5] [INFO] [1719469827.794998360] [controller_manager]: Received robot description from topic.
[ros2_control_node-5] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-5] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-5] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-5] Error parsing the limits for the interface: position from the tags [min: '{-2*pi}' and max: '{2*pi}'] within ros2_control tag inside the URDF. Skipping it
[ros2_control_node-5] Warning: Mimic joints defined in ros2_control tag are deprecated. Please define mimic joints in URDF.
[ros2_control_node-5] Warning: Mimic joints defined in ros2_control tag are deprecated. Please define mimic joints in URDF.
[ros2_control_node-5] Warning: Mimic joints defined in ros2_control tag are deprecated. Please define mimic joints in URDF.
[ros2_control_node-5] Warning: Mimic joints defined in ros2_control tag are deprecated. Please define mimic joints in URDF.
[ros2_control_node-5] Warning: Mimic joints defined in ros2_control tag are deprecated. Please define mimic joints in URDF.
[move_group-4] [INFO] [1719469827.797194525] [move_group.moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.00319745 seconds
[ros2_control_node-5] [INFO] [1719469827.797288718] [resource_manager]: Loading hardware 'KortexMultiInterfaceHardware' 
[move_group-4] [INFO] [1719469827.797424814] [move_group.moveit.moveit.core.robot_model]: Loading robot model 'gen3'...
[move_group-4] [INFO] [1719469827.797432949] [move_group.moveit.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ros2_control_node-5] [INFO] [1719469827.798029218] [resource_manager]: Loaded hardware 'KortexMultiInterfaceHardware' from plugin 'mock_components/GenericSystem'
[ros2_control_node-5] [INFO] [1719469827.798041069] [resource_manager]: Initialize hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1719469827.798057308] [resource_manager]: Successful initialization of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1719469827.798068156] [resource_manager]: Loading hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1719469827.798070907] [resource_manager]: Loaded hardware 'RobotiqGripperHardwareInterface' from plugin 'mock_components/GenericSystem'
[ros2_control_node-5] [INFO] [1719469827.798072793] [resource_manager]: Initialize hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1719469827.798080331] [resource_manager]: Successful initialization of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1719469827.798105786] [resource_manager]: 'configure' hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1719469827.798108834] [resource_manager]: Successful 'configure' of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1719469827.798111362] [resource_manager]: 'activate' hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1719469827.798112699] [resource_manager]: Successful 'activate' of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1719469827.798114127] [resource_manager]: 'configure' hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1719469827.798115191] [resource_manager]: Successful 'configure' of hardware 'KortexMultiInterfaceHardware'
[... cut ...]
[rviz2-1] [INFO] [1719469837.001531115] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1719469837.012758621] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1719469837.012907910] [rviz2.moveit.ros.motion_planning_frame]: group gripper
[rviz2-1] [INFO] [1719469837.012922506] [rviz2.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'gripper' in namespace ''
```

manadaniel, 2024-06-27:
> I confirm that I am experiencing the same. It's the same issue as [here](https://github.com/moveit/moveit2/issues/2854#issuecomment-2152224079) and this solution solved it for me.

Awesome! The proposed solution solved the problem in my case, too. Thx!

JenniferBuehler, 2024-06-27:
Glad to hear!
Remaining issue on the demo for me: I cannot see the interactive markers. I'm looking for a solution now, and will open a separate issue if there isn't any, but since you're working on this right now too I thought it's worth checking if  the markers are also not working for you?

JenniferBuehler, 2024-06-27:
Never mind I overlooked that I have to change the planning group to "manipulator". All working now :)

Leopold-Klotz, 2024-06-27:
I was not able to find success in proposed solutions above. This issue seems to better match the errors that I was working through. [Moveit2_tutorials issue 910](https://github.com/moveit/moveit2_tutorials/issues/910)

While I didn't find a solution to the specific issue, the following process ended up leading to a different result. I reinstalled from binary, as well as CycloneDDS middle ware. When following the MoveIt2 tutorials, I also had to specify `git clone --branch humble` ... when cloning the tutorial code. 
This ended up clearing the problem that I was having, hopefully it works similarly for others.

closed 2024-06-27 by Leopold-Klotz