moveit/moveit2 #2893: Binary install is not working for jazzy

celuti, 2024-06-30:
It is not possible to install MoveIt 2 from binary in Ubuntu 24.04 with ROS Jazzy: 
`sudo apt install ros-jazzy-moveit`
produces:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-jazzy-moveit`

ukicomputers, 2024-07-04:
Try with this glob, on machine that has installed ROS 2 Jazzy:
`apt-get install -y "ros-jazzy-moveit-*" --no-install-recommends`

sjahr, 2024-07-04:
The jazzy sync was delayed a bit: https://discourse.ros.org/t/preparing-for-jazzy-sync-and-patch-release-2024-06-28/38340. Does it work now?

celuti, 2024-07-04:
> The jazzy sync was delayed a bit: https://discourse.ros.org/t/preparing-for-jazzy-sync-and-patch-release-2024-06-28/38340. Does it work now?

Still not, but I managed to install each subpackage individually as proposed @ukicomputers. Thank you!

sea-bass, 2024-08-13:
Just tried

```
sudo apt list | grep ros-jazzy-moveit
```

and the packages are there now. Will go ahead and close it.

closed 2024-08-13 by sea-bass