moveit/moveit2 #2904: Moveit / setup assistant crashes when you first upload a urdf file or go to use the planning groups

LoganAnders, 2024-07-10:
### Description

Overview of your issue here.
Myself and a co-worker go to use the setup assistant mine crashes right when i upload a urdf file, his crashes right when he goes to get started either setting up planning groups or messing with the simulation itself. The main issue here being: it wont stop crashing and we dont know why.

### Your environment
* ROS Distro: humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

mikeferguson, 2024-10-03:
When I've seen this, there is usually a message in the terminal telling you that the URDF package is not found on the ROS package path.

mikeferguson, 2024-10-10:
It appears that at some time in the last few months - the crash when URDF is not on path has been fixed - I expect this issue is also fixed since we don't have any trace above

closed 2024-10-10 by mikeferguson