moveit/moveit2 #2906: [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 10.000000 seconds.

Divyam-uga, 2024-07-12:
Hello,

I have set up a Gazebo simulation with 2 UR manipulators and I'm attempting to control one UR robot using moveit_py. I've followed the necessary configuration steps, including setting use_sim_time to true in Gazebo and all related nodes. However, I'm encountering an issue where 'current_state_monitor.cpp' is not receiving joint_angles with recent timestamps.

I've verified that joint states are being published in the correct sim time using "ros2 topic echo /joint_states" and "ros2 topic echo /clock", but it seems that the requested time reflects my PC's total run time instead of the current simulation time step. I tried several ways to set the 'use_sim_time' param to true globally before initializing 'moveit_py' but it's not working out.

Any guidance on resolving this would be greatly appreciated.

Thank you.

**Error message:**
[image]

**'ur_sort.py':**
[image]

ron007d, 2024-07-31:
by adding `use_sim_time: True` in the `moveit config dict` you'll be able to get to monitor planning running but
 it will fetch another error
```console
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'qos_overrides./clock.subscription.durability' could not be set: 
Aborted (core dumped)
```

#### Python code
```python
moveit_config_dict.update({'use_sim_time' : True})
```

Jelmerdw, 2024-10-15:
I experience the same problem when using the moveit_servo servo_node in combination with Gazebo simulation. I launch all nodes with the 'use_sim_time' set to True, however the current_state_monitor still seems to use wall time. Did you found a solution?

closed 2025-01-14 by github-actions[bot]

FarStryke21, 2025-06-20:
Faced the same issue:
` Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1750447117.563005, but latest received state has time 250.872000.`

It worked when I added the use_sim_time to the moveit_config dict, 
`moveit_args["use_sim_time"] = use_sim_time`
instead of passing it as a sperate parameter to the node. 
```
parameters=[
            moveit_args,
            {"use_sim_time": use_sim_time},
        ],
```
Not sure why this works, but I would guess the MoveItPy class just takes in the one parameter and expects everything to be in there. Checking the param values for use_sim_time for the node and the internal moveitpy node tipped me off (the first one returned true and the second one was false). Hope this helps!