moveit/moveit2 #2917: No kinematics solver instantiated for group 'arm'

zhouyongyi, 2024-07-18:
### Description

Hi. I build moveit_config_package using moveit_setup_assistant from my own urdf. I install pick_ik plugin and modified kinematics.yaml as below.
```
arm:
  kinematics_solver: pick_ik/PickIkPlugin
  kinematics_solver_timeout: 0.05
  kinematics_solver_attempts: 3
  mode: global
  position_scale: 1.0
  rotation_scale: 0.5
  position_threshold: 0.001
  orientation_threshold: 0.01
  cost_threshold: 0.001
  minimal_displacement_weight: 0.0
  gd_step_size: 0.0001

```
 I write a simple node base on the instruction in [here](https://moveit.picknik.ai/main/doc/tutorials/your_first_project/your_first_project.html). When I run this simple node, I got the following error log. And real arm didn't move.

> [INFO] [1721283620.149320877] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 1.02484 seconds
[INFO] [1721283620.149397762] [moveit_robot_model.robot_model]: Loading robot model 'rml_63_gripper_description'...
[WARN] [1721283620.195971420] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[INFO] [1721283620.205834940] [move_group_interface]: Ready to take commands for planning group arm.
[INFO] [1721283620.206336457] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[INFO] [1721283621.206451246] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1721283620.206350, but latest received state has time 0.000000.
Check clock synchronization if your are running ROS across multiple machines!
[ERROR] [1721283621.206508374] [move_group_interface]: Failed to fetch current robot state

I wrote another complicated node, which control real arm to execute trajectory. I used code as below:
```
geometry_msgs::msg::Pose goal;
move_group_->setJointValueTarget(goal, "Link00");
move_group_->move();
```
After I run this code, I also get the same warning:

> [WARN] [1721283620.195971420] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!

And I got a new error log:

> [ERROR] [1721283094.638409609] [moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'arm'
[INFO] [1721283094.639067653] [move_group_interface]: Plan and Execute request accepted
[INFO] [1721283094.920604840] [move_group_interface]: Plan and Execute request complete!

But I try to get ros2 param and get result as below, it seems that I loaded kinematics solver successfully. So why these things happened?
[image]

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary install

### Expected behaviour
No Error log, and the real robot arm work as what I exepected.

### Actual behaviour
I got Error and real robot arm doesn't move at all. unless I use these kind of function:

```   
std::vector<double> goal_joints = {0.0067008, -0.422063, 1.29788, -0.032265, 1.97607, 0.0046417};
move_group_->setJointValueTarget(goal_joints);
auto res = move_group_->move();
```
If I give moveit the exact joints, the real arm will move.

navdotnetreqs, 2024-08-24:
I have the same thing running ROS2 Iron, both with gazebo and the model of a real arm. Very weirdly, however, this sample works (as the poster writes)! But just running  move_group_interface.getCurrentJointValues(); fails

https://robotics.stackexchange.com/questions/111430/failed-to-fetch-current-robot-state-when-using-mgi-in-wall-timer-callback-functi

closed 2024-11-25 by github-actions[bot]

dloret-hss, 2025-04-02:
I also have the same problem running Humble. `setPoseTarget` and `setJointValueTarget` with joint values parameter works, but `setJointValueTarget` with pose parameter does give the error:
```
[moveit_robot_state.robot_state]: No kinematics solver instantiated for group 'manipulator'
```

dloret-hss, 2025-04-03:
I just solved my issue and posting the solution in case someone else stumbles upon this. The problems seems to be that `setJointValueTarget` with a pose argument calls IK from your node, so if your node does not set the parameters in the [kinematics.yaml](https://moveit.picknik.ai/humble/doc/examples/kinematics_configuration/kinematics_configuration_tutorial.html) configuration file, it cannot instantiate the IK solver. When I loaded the contents of the kinematics file as parameters to my node, I was able to call `setJointValueTarget` with pose arguments and plan in joint space.

lnss1-ua, 2025-05-12:
> I just solved my issue, and I am posting the solution in case someone else stumbles upon this. The problem seems to be that `setJointValueTarget` with a pose argument calls IK from your node, so if your node does not set the parameters in the [kinematics.yaml](https://moveit.picknik.ai/humble/doc/examples/kinematics_configuration/kinematics_configuration_tutorial.html) configuration file, it cannot instantiate the IK solver (even though the `move_group` node loads the file). When I loaded the contents of the kinematics file as parameters to my node, I was able to call `setJointValueTarget` with pose arguments and plan in joint space.

can i ask you how did you manage to load the contents of the kinematics as parameters? I tried: 
nodo_->declare_parameter("lite6.kinematics_solver", "trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin");
        nodo_->declare_parameter("lite6.kinematics_solver_search_resolution", 0.005);
        nodo_->declare_parameter("lite6.kinematics_solver_timeout", 0.005);
        nodo_->declare_parameter("lite6.kinematics_solver_attempts", 3);

but doesn't work