moveit/moveit2 #2919: Error building robotiq_driver

marlow-fawn, 2024-07-18:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build
* If binary, which release version?

### Steps to reproduce
Followed tutorial for [Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) up to colcon build.

### Expected behaviour
robotiq_driver package builds

### Actual behaviour
robotiq_driver package fails to build

Seems like it could be a version mismatch of the ResourceManager? Candidates don't seem aligned.

### Backtrace or Console output

[Stack Trace](https://gist.github.com/marlow-fawn/37cbc0ade9a31fc1b5bcef05c3e8dde5)

sea-bass, 2024-08-03:
The main branch of that repo is incompatible with Humble since this PR: https://github.com/PickNikRobotics/ros2_robotiq_gripper/pull/54

You will need to grab the `humble` branch instead and that should fix the error.

sjahr, 2024-09-17:
Closing this as it looks like it is resolved

closed 2024-09-17 by sjahr