moveit/moveit2 #2928: Request for a new release of MoveIt into Rolling

clalancette, 2024-07-24:
### Description

Following up on https://github.com/moveit/moveit2/pull/2881 , it would be great to get a new release of MoveIt into Rolling.  Once that is done, we'll be able to remove the octomap package from Rolling itself, and thus be able to close https://github.com/ros/rosdistro/issues/41622

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 24.04
* Binary
* New one, please :).
* Which RMW (Fast DDS or Cyclone DDS)? N/A

clalancette, 2024-07-31:
Friendly ping here; this is the last remaining package before we can remove `octomap` from Rolling.

clalancette, 2024-08-23:
Just as an FYI, I've now opened up https://github.com/ros/rosdistro/pull/42561 .  Once that is approved and merged, Moveit will no longer compile in Rolling until a new release is made.

rhaschke, 2024-08-30:
@sjahr, @henningkayser: Is there any reason (except time constraints) that you didn't cut a new release yet?
If time is the only constraint, I can offer to create a new release.

sjahr, 2024-09-16:
@rhaschke @clalancette Sorry I completely lost track of this and was out of office the past two weeks. Yes, at my end, time constraints are the only issue blocking this :see_no_evil: @rhaschke If you have the capacities to cut a new release, that would be awesome

rhaschke, 2024-09-16:
Will do.

rhaschke, 2024-09-16:
Release prepared: https://github.com/moveit/moveit2/pull/3002
But I'm lacking direct push permissions required by the release script.

rhaschke, 2024-09-25:
Done: #3002

closed 2024-09-25 by rhaschke