moveit/moveit2 #2935: setup_assistant launch files don't work

Karlosbubi, 2024-07-26:
### Description

I've used the setup_assistant to create a moveit_config package from my urdf file (of a meca500), but I can't launch anything.
I'm pretty certain the urdf file is fine, since the assistant doesn't complain and it worked in ROS1.
One of the problems i encountered is this one  #1814 i've used the workaround mentioned there, that might impact the other results though.

### Your environment
* ROS Distro: Humble (& Jazzy)
* OS Version: Ubuntu 22.04 (& 24.04)
* Source or Binary build?: Source Build (%& binary build)
* If binary, which release version?: 2.10.0-1noble.20240719.210749
* If source, which branch? : Iron
* Which RMW (Fast DDS or Cyclone DDS)? : ROS2 default

* I mostly work with humble, but due to the lackluster support for python i also did some test with jazz to see if it was fixed in the mean time (as far as i can tell it wasn't)

### Steps to reproduce

- install moveit2
- get an URDF file
- launch setup assiastan `ros2 launch moveit_setup_assistant setup_assistant.launch.py``
- launch the generated launch files
  - e.g. `ros2 launch meca_moveit2_config demo.launch.py `
  - e.g. `ros2 launch meca_moveit2_config move_group.launch.py `

### Expected behaviour

RViz should lauch with the robot model and the moveit nodes should be spawned

### Actual behaviour

RViz launches without the model and the moveit nodes do not appear

### Backtrace or Console output

In case of `ros2 launch meca_moveit2_config demo.launch.py ` : 
```
[rviz2-4] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-4]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
[spawner-7] [INFO] [1721997461.939915429] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' node to exist
[ros2_control_node-5] [INFO] [1721997462.153337706] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-7] [INFO] [1721997462.215030818] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-5] [INFO] [1721997462.216510785] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-5] [INFO] [1721997462.216600127] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-7] [INFO] [1721997462.514558611] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-7]: process has finished cleanly [pid 240080]
[rviz2-4] [ERROR] [1721997464.218080710] [moveit_2326694259.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-4] [INFO] [1721997464.254434545] [moveit_2326694259.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-4] [ERROR] [1721997474.520481676] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
[rviz2-4] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-4]          at line 715 in ./src/model.cpp
[rviz2-4] [ERROR] [1721997474.542100358] [moveit_2326694259.moveit.ros.rdf_loader]: Unable to parse SRDF
[rviz2-4] [ERROR] [1721997474.565024350] [moveit_2326694259.moveit.ros.planning_scene_monitor]: Robot model not loaded
```

In case of `ros2 launch meca_moveit2_config move_group.launch.py ` :
```
[move_group-1] [INFO] [1721997672.534403688] [move_group.moveit.moveit.ros.rdf_loader]: Loaded robot model in 0.033171 seconds
[move_group-1] [INFO] [1721997672.534762827] [move_group.moveit.moveit.core.robot_model]: Loading robot model 'meca_500_r3'...
[move_group-1] [INFO] [1721997672.575961874] [move_group.moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'meca_arm': 1 1 1 1 1 1
[move_group-1] [INFO] [1721997672.576678089] [move_group.moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'meca': 1 1 1 1 1 1
[move_group-1] Stack trace (most recent call last):
[move_group-1] #14   Object "", at 0xffffffffffffffff, in 
[move_group-1] #13   Object "/home/***/meca_ros2_workspace/build/moveit_ros_move_group/move_group", at 0x55c657191094, in _start
[move_group-1] #12   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9b3a50de3f, in __libc_start_main
[move_group-1] #11   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f9b3a50dd8f, in 
[move_group-1] #10   Object "/home/***/meca_ros2_workspace/build/moveit_ros_move_group/move_group", at 0x55c65719021e, in main
[move_group-1] #9    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.10.0", at 0x7f9b3ae3ad5a, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-1] #8    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.10.0", at 0x7f9b3ae3a71b, in moveit_cpp::MoveItCpp::loadPlanningSceneMonitor(moveit_cpp::MoveItCpp::PlanningSceneMonitorOptions const&)
[move_group-1] #7    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.10.0", at 0x7f9b3ad41231, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #6    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.10.0", at 0x7f9b3ad411a6, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #5    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.10.0", at 0x7f9b3ad40d74, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::shared_ptr<robot_model_loader::RobotModelLoader> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #4    Object "/home/***/meca_ros2_workspace/install/moveit_ros_planning/lib/libmoveit_planning_scene_monitor.so.2.10.0", at 0x7f9b3ad3fa84, in planning_scene_monitor::PlanningSceneMonitor::initialize(std::shared_ptr<planning_scene::PlanningScene> const&)
[move_group-1] #3    Object "/home/***/meca_ros2_workspace/install/moveit_core/lib/libmoveit_planning_scene.so.2.10.0", at 0x7f9b3a092bd5, in planning_scene::PlanningScene::PlanningScene(std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<collision_detection::World> const&)
[move_group-1] #2    Object "/home/***/meca_ros2_workspace/install/moveit_core/lib/libmoveit_planning_scene.so.2.10.0", at 0x7f9b3a09206e, in planning_scene::PlanningScene::initialize()
[move_group-1] #1    Object "/home/***/meca_ros2_workspace/install/moveit_core/lib/libmoveit_robot_state.so.2.10.0", at 0x7f9b3a3f2945, in moveit::core::RobotState::~RobotState()
[move_group-1] #0    Object "/home/***/meca_ros2_workspace/install/moveit_core/lib/libmoveit_robot_state.so.2.10.0", at 0x7f9b3a3f283a, in moveit::core::RobotState::clearAttachedBodies()
[move_group-1] Segmentation fault (Address not mapped to object [0x40])
[ERROR] [move_group-1]: process has died [pid 241068, exit code -11, cmd '/home/***/meca_ros2_workspace/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_t_2_4ps_ --params-file /tmp/launch_params_t_e0684e'].
```

* These outputs are on humble but they are pretty identical on jazzy exept for the cases i run into `[ERROR] [launch]: Caught exception in launch (see debug for traceback): 'capabilities'` on jazzy cause there i have the binary package and can't easily apply the workaround

Karlosbubi, 2024-07-29:
I've graduated to the next wierd error message
`[move_group-1] terminate called after throwing an instance of 'pluginlib::LibraryLoadException'
[move_group-1]   what():  Failed to load library /opt/ros/jazzy/lib/libsdformat_urdf_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libsdformat14.so.14: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99` from `ros2 launch meca_moveit2_config move_group.launch.py` (source build main branch on jazzy)

closed 2024-07-30 by Karlosbubi

rawkss, 2024-07-30:
> ### Description
> I've used the setup_assistant to create a moveit_config package from my urdf file (of a meca500), but I can't launch anything. I'm pretty certain the urdf file is fine, since the [...]

Hi, how did you fix the (Unable to parse SRDF error) exactly? I am getting the same error with Jazzy on Ubuntu 24.04.

Karlosbubi, 2024-08-14:
> With some Voodoo everything works now (except the "capabilities" thing). Unfortunately I don't quite understand and therefore can't tell you how exactly to solve it poor soul who googles the problem next.

I didn't it fixed itself or something