moveit/moveit2 #2940: moveit_py -> use_sim_time: True fetching Error rclcpp::exceptions::InvalidParameterValueException

ron007d, 2024-07-31:
### Description
Using panda_moveit_config from moveit resources, 
added another launch file to launch the robot in gazebo.
- Gazebo model loaded  = :white_check_mark: 
- rviz loaded = :white_check_mark: 
- Rvix motion planning able to control = :white_check_mark: 
- moveit py = :negative_squared_cross_mark: 

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
*  Source
* main

### Steps to reproduce
1. clone this package > moveit resources - [https://github.com/moveit/moveit_resources](https://github.com/moveit/moveit_resources)
```console
git clone https://github.com/moveit/moveit_resources
```
2. in the panda_moveit_config package create a file name demo_gazebo.launch.py and copy and paste this python code
```python
import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument,IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration, PathJoinSubstitution
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.conditions import IfCondition
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare
from ament_index_python.packages import get_package_share_directory,get_package_prefix
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():
    os.environ['GAZEBO_MODEL_PATH'] =f"{get_package_prefix('moveit_resources_panda_description')}/share/:{os.environ['GAZEBO_MODEL_PATH']}" 
    # print(os.environ['GAZEBO_MODEL_PATH'])

    gazebo = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([os.path.join(
            get_package_share_directory('gazebo_ros'), 'launch'), '/gazebo.launch.py']),
        )
    
    spawn_entity = Node(package='gazebo_ros', executable='spawn_entity.py',
                    arguments=['-topic', 'robot_description',
                                '-entity', 'panda', '-x', '0.0','-y', '0.0','-z', '0.0'],
                    output='screen')
    
    # Command-line arguments
    rviz_config_arg = DeclareLaunchArgument(
        "rviz_config",
        default_value="moveit.rviz",
        description="RViz configuration file",
    )

    db_arg = DeclareLaunchArgument(
        "db", default_value="False", description="Database flag"
    )

    ros2_control_hardware_type = DeclareLaunchArgument(
        "ros2_control_hardware_type",
        default_value="mock_components",
        description="ROS 2 control hardware interface type to use for the launch file -- possible values: [mock_components, isaac]",
    )

    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_panda")
        .robot_description(
            file_path="config/panda.gazebo.urdf",
        )
[... 214 lines cut ...]

    plan_and_execute(panda, panda_arm, logger, sleep_time=3.0)

if __name__ == '__main__':
    main()
```

### Expected behaviour
Robot moving to extented position

### Actual behaviour
python code error, kernel crash in jupyter notebook

### Console output
```console
[INFO] [1722406022.160341719] [moveit_3747324773.moveit.plugins.simple_controller_manager]: Added FollowJointTrajectory controller for panda_arm_controller
[INFO] [1722406022.160445906] [moveit_3747324773.moveit.plugins.simple_controller_manager]: Max effort set to 0.0
[INFO] [1722406022.163541288] [moveit_3747324773.moveit.plugins.simple_controller_manager]: Added GripperCommand controller for panda_hand_controller
[INFO] [1722406022.163789564] [moveit_3747324773.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[INFO] [1722406022.163845204] [moveit_3747324773.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[INFO] [1722406022.164344371] [moveit_3747324773.moveit.ros.trajectory_execution_manager]: Trajectory execution is managing controllers
terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
  what():  parameter 'qos_overrides./clock.subscription.durability' could not be set: 
Aborted (core dumped)
```

ra-georgi, 2024-10-07:
Facing the same issue

ron007d, 2024-10-09:
> Facing the same issue

this helped me
```python
config_dict = moveit_config.to_dict()
config_dict.update({'use_sim_time':True})
```

ogios, 2024-11-07:
> this helped me
> 
> config_dict = moveit_config.to_dict()
> config_dict.update({'use_sim_time':True})

@ron007d 
what does this mean? isn't this the cause of the issue?
i have the same problem here with this value set

ron007d, 2024-11-07:
> > this helped me
> > config_dict = moveit_config.to_dict()
> > config_dict.update({'use_sim_time':True})
> 
> [@ron007d](https://github.com/ron007d) what does this mean? isn't this the cause of the issue? i have the same problem here with this value set

@ogios I had this error and what I found that its not subscribed to /clock

then updating this key in dictionary the moveit node subscribed to the /clock

ogios, 2024-11-08:
alright. i used to directly run the script inside console.
but now, i created a new launch file to include this script, and move moveit_config into the launch file, pass to the script as the parameter.
and it worked...

I don't understand why

ogios, 2024-11-08:
Ah! Found one better solution.

When i'm using the launch file, i notice the argument passed to the script: `--params-file /tmp/launch_params_xp_ss6gq`.
And as i go though MoveItPy cpp code [here](https://github.com/moveit/moveit2/blob/1a77d7d5154108b6ad0681f17502092a0b374a5e/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L63), i found it containing lots of `launch_params` related stuff which also includes an argument: `--params-file`.

The logic writes either `config_dict` or `launch_params_filepaths`, and they all end up with `--params-file` none the less. just that `config_dict` is converted into yaml by `moveit::utils::create_params_file_from_dict`.

So i compare two files, and found the one major difference is that the node_name at the root, the one from launch file is `/**`, and the one from `config_dict` is the node_name i passed into `MoveItPy`.
And i try to convert the dict outside of `MoveItPy` and pass in as `launch_params_filepaths`:
```python
moveit_config.update({"use_sim_time": True})
file = create_params_file_from_dict(moveit_config, "/**")
arm = MoveItPy(
    node_name=node_name,
    launch_params_filepaths=[file],
)
```

And it worked, just simple as that.

but idk why node name causes this to happen...

closed 2025-02-07 by github-actions[bot]