moveit/moveit2 #2947: typo "trajactory" in moveit_py

christian-rauch, 2024-08-05:
### Description

There is a typo "traj**a**ctory" -> "traj**e**ctory" in `moveit_py`:
https://github.com/moveit/moveit2/blob/19d4a2818bca7024305cbf5be5c3bca4186351d5/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L174-L178

Unfortunately, fixing this typo will break API.

### Expected behaviour
The method should be named `get_trajectory_execution_manager`.

### Actual behaviour
The method is incorrectly named `get_trajactory_execution_manager`.

sjahr, 2024-08-05:
@christian-rauch Thanks for reporting this. Would you mind opening a PR to fix it?

akky20, 2024-08-06:
@sjahr I have raised a PR for this, can you please have a look into it

closed 2024-08-06 by sjahr

JensVanhooydonck, 2024-10-12:
I guess there is still a typo in the pyi file: https://github.com/moveit/moveit2/blob/865ad364e1a8ae1582f9dfb8428eb4c152f3aa10/moveit_py/moveit/planning.pyi#L19

PR #2951 (2024-08-06): fixed typo from trajactory to trajectory
### Description

Fixes #2947 

Edited the typo in line from "trajactory" to "trajectory"

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/moveit/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [reference](https://moveit.picknik.ai/humbl [... cut]