moveit/moveit2 #2955: Servo cannot reach the desired control rate

AndyZe, 2024-08-07:
### Description

Launch a Servo demo, e.g.

`ros2 launch moveit_servo demo_ros_api.launch.py`

`ros2 run moveit_servo servo_keyboard_input`

Monitor the joint command publication rate:

`ros2 topic hz -w 500 /panda_arm_controller/joint_trajectory`

It's nowhere close to the desired rate of 100 Hz. I get ~60 Hz. Not sure exactly when this regression happened but I suspect PR #2651.

AndyZe, 2024-08-07:
It looks like changing to the `online_signal_smoothing::ButterworthFilterPlugin` instead of `online_signal_smoothing::AccelerationLimitedPlugin` does help, it increases the control rate to ~75 Hz. But that's still lower than the requested 100Hz, which used to be easy to achieve. Previously 1,000 Hz was achievable.

sea-bass, 2024-08-08:
I see a lower rate when I try to toggle back and forth between commands because there is some dead time in between, but if I just hold down a button and go right into joint limits/singularity, I hit 100 Hz no problem on my 4 year old laptop?

https://github.com/user-attachments/assets/cd3a77b2-6482-47fb-9d7c-c0e4a440cd7a

sea-bass, 2024-08-08:
Here's another one in which I am just spamming arrow keys in twist mode.

https://github.com/user-attachments/assets/875d9217-620a-4c5b-81af-47c70074c50b

closed 2024-08-08 by AndyZe