moveit/moveit2 #2972: Change the starting point of planning under the plug-in using a topic or service

MrVBian, 2024-08-15:
Hello Everyone,

I need to be able to change the starting point of planning under the MotionPlanning plugin via topic or service. But I don't know how to do it.

Any guidance on resolving this would be greatly appreciated.

Thank you.

MrVBian, 2024-08-15:
Add a Topic Monitor picture:

[image]

sjahr, 2024-08-15:
Please take a look at the [tutorials](https://moveit.picknik.ai/main/index.html), especially the ones around the move_group interface. For future questions on how to use moveit, please use https://robotics.stackexchange.com/

closed 2024-08-15 by sjahr

MrVBian, 2024-08-15:
> Please take a look at the [tutorials](https://moveit.picknik.ai/main/index.html), especially the ones around the move_group interface. For future questions on how to use moveit, please use https://robotics.stackexchange.com/

Thank you for your reply. @sjahr 

Sorry, after reading the document, I still can't figure out how to change the starting point of the plan through topic or service. Can you give me some more Pointers. For example, which topic or service should I study. 

Thanks again for your reply.

MrVBian, 2024-08-15:
[https://robotics.stackexchange.com/](https://robotics.stackexchange.com/questions/112648/how-to-change-the-starting-point-of-planning-under-the-plug-in-using-a-topic-or). This question was also asked

sjahr, 2024-08-15:
The [MotionPlanRequest has a field called start_state](https://github.com/moveit/moveit_msgs/blob/9756797dcbc293c7cae66541a7bc0c7acec85892/msg/MotionPlanRequest.msg#L10) you can use to set the start state. The motion plan request can be set either with the move_group interface or moveit_cpp

MrVBian, 2024-08-15:
I think you misunderstood my question.
- The setStartState function in move_group_interface cannot modify the start attitude in rviz. The same cannot be said for MotionPlanRequest.
- What I need is not a custom start point and end point, and then solve a trajectory.

What I need is how to change the green Query start state under the moveit plugin after rviz is started.

[image]

rhaschke, 2024-08-15:
To enable external control of the rviz MotionPlanning plugin, you need to tick the corresponding checkbox first:
[image]

Then, you can send start / goal states to the topics `/rviz/moveit/update_custom_start_state` and `/rviz/moveit/update_custom_goal_state`. 
The topics `/rviz/moveit/update_start_state` and `/rviz/moveit/update_goal_state`, accepting an empty message, reset the corresponding state to the current one.

MrVBian, 2024-08-16:
Thank you for your reply, which answered my question. @rhaschke