moveit/moveit2 #2980: Moveit Setup Assistant crashes in Robot Poses tab

shushutunn, 2024-08-25:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If source, which branch? humble
* Which RMW (Fast DDS or Cyclone DDS)?  Cyclone DDS

### Steps to reproduce

1. Launching moveit setup assistant.
`ros2 launch moveit_setup_assistant setup_assistant.launch.py`
2. Click "Create New Moveit Configuration Package"
3.  Click "Browse" and select `panda.urdf`.
4. Click "Load Files"
5. In Self-Collisions,set Sampling Density as 50000, then click "Generate Collision Matrix"
6. Set virtual joint
[image]
7.  Create panda_arm Planning Group
[image]
8. When move a sliderin Robot Poses tab, suddenly the software crashes.
[image]

### Expected behaviour
Same behavior as this tutorial, with no crash.
https://moveit.picknik.ai/humble/doc/examples/setup_assistant/setup_assistant_tutorial.html

### Actual behaviour
Suddenly crash when move a slider. The window closes soon.

### Backtrace or Console output
https://gist.github.com/shushutunn/f614ccceb97af377ae960ae3009d1e39

JiangShangJiu, 2024-09-05:
I'm having the same issue, do you already have a workaround

WikiGenius, 2024-09-05:
I'm having the same issue

shushutunn, 2024-09-06:
Not yet.

shushutunn, 2024-09-06:
> I'm having the same issue, do you already have a workaround

> I'm having the same issue, do you already have a workaround

Not yet

WikiGenius, 2024-09-06:
it worked  
follow this

> I have solved uninstalling moveit2 binaries with 'sudo apt remove ros-humble-moveit*'
> 
> Then deleting the workspace folder, do again the Getting started tutorials.
> 
> The issue was created for some conflict with binaries with respect to the build of the workspace. it seems you can't have both as for ROS1

https://github.com/moveit/moveit2/issues/2248#issuecomment-1611896152

mikeferguson, 2024-10-21:
This really does appear to be ABI incompatibility - uninstalling the apt packages is a temporary workaround - the real fix would be for us to put a new release out - per https://github.com/moveit/moveit2/issues/3012

rayee922, 2025-07-15:
> This really does appear to be ABI incompatibility - uninstalling the apt packages is a temporary workaround - the real fix would be for us to put a new release out - per [#3012](https://github.com/moveit/moveit2/issues/3012)

Hi,

I am using the new release. And I also run 'sudo apt remove ros-humble-moveit*' before build from source. But Moveit Setup Assistant still crashed when move a slider in Robot Poses tab. Do you have any suggesstions? Thank you so much.