moveit/moveit2 #2986: Adding MoveIt2 #include's stop VS Code Intellisense from working

navdotnetreqs, 2024-08-31:
### Description

Adding MoveIt2 #include-line(s) to C++ project stops VS Code IntelliSense from working.

- #include <moveit/move_group_interface/move_group_interface.h>
or
- #include <moveit/move_group/move_group_capability.h>

Adding other libraries or #include <moveit/move_group/capability_names.h> does not break functionality.

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04
* Binary build
* ros-iron-moveit 2.8.0-1jammy.20240815.154016 amd64
* Default RMW

### Steps to reproduce

After having set up VS Code with proper extensions (c/c++, ros, cmake etc), I can confirm IntelliSense is working both with error squiggles and autocomplete for a test c++-file.

[image]
```
#include <cstdio>
#include <rclcpp/rclcpp.hpp>

int main(int argc, char ** argv)
{
  (void) argc;
  (void) argv;
  rclcpp::init(argc, argv);
auto const node = std::make_shared<rclcpp::Node>(
    "hello_moveit",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );
    
  auto const logger = rclcpp::get_logger("hello_moveit");

  // Next step goes here

  // Shutdown ROS
  rclcpp::shutdown();

  printf("hello world hello_moveit package\n");
  return 0;
}
```

As soon as I add #include for movit (from tutorial)

#include <moveit/move_group_interface/move_group_interface.h>

IntelliSense stops working completely. I get no include errors. Compiles ok with colcon.

[image]

### Expected behaviour
I would expect IntelliSense to keep working, and allow me error squiggles and autocompletion for objects from the MoveIt-library as well as all others.

### Actual behaviour
No errors are shown, nothing is autocompleted anymore until I remove the include.

### Backtrace or Console output

Same thing happens even if moveit-#include is included from another header-file (hello_moveit.h)
I can autocomplete stuff from stdio, but if I add #include "hello_moveit.h" (which includes move_group_interface.h), IntelliSense stops again.
[image]

With move_group/capability_names.h it works, this would suggest that library include paths are ok.
[image]

Any ideas?

rhaschke, 2024-09-02:
This seems to be a bug in VS Code, not MoveIt. I guess, including `moveit/move_group_interface/move_group_interface.h` just requires Code to perform analysis of this code, which takes a while. During this time, IntelliSense is not available?
For me, IntelliSense works in the MoveIt code base.

navdotnetreqs, 2024-09-02:
Hi @rhaschke, you are probably right, closing this. :)

closed 2024-09-02 by navdotnetreqs