moveit/moveit2 #2987: Humble MoveIt2 Moveit_servo

kavikode, 2024-09-02:
moveit2/moveit_ros/moveit_servo/test/servo_launch_test_common.hpp:54:10: fatal error: moveit_msgs/srv/change_drift_dimensions.hpp: No such file or directory
   54 | #include <moveit_msgs/srv/change_drift_dimensions.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

ROS2 Humble on Ubuntu 22.04

even though I used 
git clone -b humble https://github.com/moveit/moveit2_tutorials
vcs import --recursive < moveit2_tutorials/moveit2_tutorials.repos
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
cd ..
source /opt/ros/humble/setup.bash
colcon build --mixin release --executor sequential

It still does not compile in order to use motion planning API in C++.

Please provide assistance.
Thank you

AndyZe, 2024-09-02:
Check if you're on the `humble` branch of moveit_msgs. The missing service appears to be there: 

https://github.com/moveit/moveit_msgs/blob/humble/srv/ChangeDriftDimensions.srv

kavikode, 2024-09-03:
thank you so much! It seems the following was necessary:

git clone --branch humble https://github.com/ros-planning/moveit2_tutorials.git
git clone https://github.com/moveit/moveit2.git --branch humble

sea-bass, 2024-09-16:
Thanks for confirming!

closed 2024-09-16 by sea-bass