moveit/moveit2 #2989: RViz ignores 'joint_limits.yaml' in MoveIt Config when using moveit_rviz default launch file

Dabuss, 2024-09-03:
### Description: 
I encountered an issue where RViz ignores the joint limits specified in the joint_limits.yaml file within the MoveIt configuration package when launching using the default launch files generated by the MoveIt2 Setup Assistant.
Steps to Reproduce:

- Create a MoveIt2 configuration package using the Setup Assistant.
- Define custom joint limits in the joint_limits.yaml file within the config directory.
- Notice how these joint_limits are ignored by looking at rviz parameters

### Observed Behavior:
In my case, Motion planning within RViz results in unexpectedly large joint movements. Upon further inspection, it appears that RViz is using the joint limits from the URDF (in my case, a UR10e robot with 720° angle ranges), rather than the limits specified in joint_limits.yaml present in the moveit_config of my robot package.

### Expected Behavior:
RViz should respect and use the joint limits defined in the joint_limits.yaml file when performing motion planning.

### My Solution:
I modified the generate_moveit_rviz_launch function to include the joint limits in the node parameters passed to RViz. Here is the reimplementation of the function:

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from moveit_configs_utils import MoveItConfigsBuilder

def generate_moveit_rviz_launch_using_joint_limits(moveit_config):
    """Launch file for RViz with joint limits"""
    ld = LaunchDescription()

    ld.add_action(DeclareBooleanLaunchArg("debug", default_value=False))
    ld.add_action(
        DeclareLaunchArgument(
            "rviz_config",
            default_value=str(moveit_config.package_path / "config/moveit.rviz"),
        )
    )

    rviz_parameters = [
        moveit_config.planning_pipelines,
        moveit_config.robot_description_kinematics,
        moveit_config.robot_description_joint_limits  # Added joint limits here
    ]

    add_debuggable_node(
        ld,
        package="rviz2",
        executable="rviz2",
        output="log",
        respawn=False,
        arguments=["-d", LaunchConfiguration("rviz_config")],
        parameters=rviz_parameters,
    )

    return ld
```

### Question:

Is this behavior intentional, or is it an oversight that RViz does not use the joint limits from joint_limits.yaml? Should the default generate_moveit_rviz_launch function be updated to include these limits, or is there another recommended approach?

Thank you for your assistance!

m-elwin, 2024-11-14:
I was having this same issue and created a PR #3901. My solution was basically the same as @Dabuss although it seems like `robot_description_joint_limits` became just `joint_limits` at some point.

mikeferguson, 2024-11-27:
This is fixed in main - I just started the back port to humble in https://github.com/moveit/moveit2/pull/3136

closed 2024-11-27 by mikeferguson