moveit/moveit2 #3004: Broken version.h file

saikishor, 2024-09-18:
### Description

```
root@417ad4e895ce:/# cat /opt/ros/rolling/include/moveit_core/moveit/version.h 

/*********************************************************************
* Software License Agreement (BSD License)
*
*  Copyright (c) 2013, Willow Garage, Inc.
*  All rights reserved.
*
*  Redistribution and use in source and binary forms, with or without
*  modification, are permitted provided that the following conditions
*  are met:
*
*   * Redistributions of source code must retain the above copyright
*     notice, this list of conditions and the following disclaimer.
*   * Redistributions in binary form must reproduce the above
*     copyright notice, this list of conditions and the following
*     disclaimer in the documentation and/or other materials provided
*     with the distribution.
*   * Neither the name of the Willow Garage nor the names of its
*     contributors may be used to endorse or promote products derived
*     from this software without specific prior written permission.
*
*  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
*  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
*  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
*  FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
*  COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
*  INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
*  BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
*  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
*  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
*  LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
*  ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
*  POSSIBILITY OF SUCH DAMAGE.
*********************************************************************/

#pragma once

#define MOVEIT_VERSION_STR "..-devel"

#define MOVEIT_VERSION_MAJOR 
#define MOVEIT_VERSION_MINOR 
#define MOVEIT_VERSION_PATCH 

#define MOVEIT_GIT_NAME ""
#define MOVEIT_GIT_COMMIT_HASH ""

/// MOVEIT_VERSION is (major << 16) + (minor << 8) + patch.
#define MOVEIT_VERSION MOVEIT_VERSION_CHECK(MOVEIT_VERSION_MAJOR, MOVEIT_VERSION_MINOR, MOVEIT_VERSION_PATCH)

/// Use like: #if MOVEIT_VERSION >= MOVEIT_VERSION_CHECK(1, 0, 0)
#define MOVEIT_VERSION_CHECK(major, minor, patch) ((major << 16) | (minor << 8) | (patch))
```

it seems like the `version.h` is not properly defined at all. I believe it should have the proper numbers from the respective tags.

The issue is observed in Iron, Rolling as well as jazzy. I suspect the problem is after this change: https://github.com/moveit/moveit2/commit/f6d9aa49f1b1270d89d50d89a6f31f7ce964ba23

One more thing, the `verison.h` in Humble is placed in `/opt/ros/humble/include/moveit/version.h` and the one of rolling and iron are placed in `moveit_core/moveit/version.h` , I think it is better to unify it for better reusing between distros

### ROS Distro

Iron

### OS and version

Ubuntu 22.04

### Source or binary build?

Binary

### If binary, which release version?

2.8.0 (The issue could be from 2.7.0)

### Which RMW are you using?

None

### Steps to Reproduce

cat /opt/ros/rolling/include/moveit_core/moveit/version.h 

### Expected behavior

Properly defined 

```
#define MOVEIT_VERSION_MAJOR 2
#define MOVEIT_VERSION_MINOR 10
#define MOVEIT_VERSION_PATCH 0

### Actual behavior

```
#define MOVEIT_VERSION_MAJOR 
#define MOVEIT_VERSION_MINOR 
#define MOVEIT_VERSION_PATCH

saikishor, 2024-11-04:
@henningkayser @sjahr or any other maintainer 
Can you please confirm the issue?

rr-tom-noble, 2024-11-19:
This [seems to have been fixed in the source](https://github.com/moveit/moveit2/blob/iron/moveit_core/version/version.h.in). I wonder if there simply isn't an up-to-date iron binary release? I believe it's reaching EOL very soon.

henningkayser, 2025-03-25:
Iron has reached eol in December. Please migrate to Jazzy or Rolling

saikishor, 2025-03-25:
@henningkayser This issue is not solved at all

```
cat /opt/ros/rolling/include/moveit_core/moveit/version.hpp 
/*********************************************************************
* Software License Agreement (BSD License)
*
*  Copyright (c) 2013, Willow Garage, Inc.
*  All rights reserved.
*
*  Redistribution and use in source and binary forms, with or without
*  modification, are permitted provided that the following conditions
*  are met:
*
*   * Redistributions of source code must retain the above copyright
*     notice, this list of conditions and the following disclaimer.
*   * Redistributions in binary form must reproduce the above
*     copyright notice, this list of conditions and the following
*     disclaimer in the documentation and/or other materials provided
*     with the distribution.
*   * Neither the name of the Willow Garage nor the names of its
*     contributors may be used to endorse or promote products derived
*     from this software without specific prior written permission.
*
*  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
*  "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
*  LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
*  FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
*  COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
*  INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
*  BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
*  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
*  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
*  LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
*  ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
*  POSSIBILITY OF SUCH DAMAGE.
*********************************************************************/

#pragma once

#define MOVEIT_VERSION_STR "..-devel"

#define MOVEIT_VERSION_MAJOR 
#define MOVEIT_VERSION_MINOR 
#define MOVEIT_VERSION_PATCH 

#define MOVEIT_GIT_NAME ""
#define MOVEIT_GIT_COMMIT_HASH ""

/// MOVEIT_VERSION is (major << 16) + (minor << 8) + patch.
#define MOVEIT_VERSION MOVEIT_VERSION_CHECK(MOVEIT_VERSION_MAJOR, MOVEIT_VERSION_MINOR, MOVEIT_VERSION_PATCH)

/// Use like: #if MOVEIT_VERSION >= MOVEIT_VERSION_CHECK(1, 0, 0)
#define MOVEIT_VERSION_CHECK(major, minor, patch) ((major << 16) | (minor << 8) | (patch))

```

This is with the moveit_core version in the rolling 2.13.0

saikishor, 2025-03-25:
It would be great if someone can take a look into it

saikishor, 2025-04-02:
@sjahr FYI