moveit/moveit2 #3005: ServoCppFixture.PoseTest is flaky

rhaschke, 2024-09-18:
### Description

failure: https://github.com/moveit/moveit2/actions/runs/10915496106/job/30295360965
success: https://github.com/moveit/moveit2/actions/runs/10915496106

### ROS Distro

Rolling

### OS and version

Ubuntu 24.04

### Source or binary build?

Source

### If source, which branch?

main

### Which RMW are you using?

None

### Steps to Reproduce

run CI actions

### Expected behavior

tests should always succeed

### Actual behavior

sometimes the test fails

sea-bass, 2024-09-20:
The "bad" status code for this test indicates that the robot is halting for collision.

I guess picking a safer target pose could be helpful, but I think using a full IK solver instead of something simpler/more conducive to online control like a Jacobian pseudoinverse could also be a cause.

sea-bass, 2024-09-20:
Attempting something in https://github.com/moveit/moveit2/pull/3007 -- but unclear if this will help.

sea-bass, 2024-09-20:
> Attempting something in #3007 -- but unclear if this will help.

Just caught the flake again in this PR. Sadly, it does not fix it. Let's keep this issue open.

rhaschke, 2024-10-06:
I just saw another test failure: https://github.com/moveit/moveit2/actions/runs/11203600643/job/31141038313?pr=3020

mikeferguson, 2024-10-29:
Copying the text here - since the github CI runs eventually disappear: 

```
    ======================================================================
    FAIL: test_gtest_pass (moveit_servo.TestGTestProcessPostShutdown.test_gtest_pass)
    ----------------------------------------------------------------------
    Traceback (most recent call last):
      File "/home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/moveit_servo/tests/launch/servo_cpp_integration.test.py", line 133, in test_gtest_pass
        launch_testing.asserts.assertExitCodes(proc_info, process=servo_gtest)
      File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
        assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    AssertionError: Proc moveit_servo_cpp_integration_test-5 exited with code 1
    
    ----------------------------------------------------------------------
    Ran 1 test in 0.001s
    
    FAILED (failures=1)
    -- run_test.py: return code 1
    -- run_test.py: verify result file '/home/runner/work/moveit2/moveit2/.work/target_ws/build/moveit_servo/test_results/moveit_servo/tests_launch_servo_cpp_integration.test.py.xunit.xml'
  >>>
build/moveit_servo/test_results/moveit_servo/tests_launch_servo_cpp_integration.test.py.xunit.xml: 2 tests, 0 errors, 1 failure, 0 skipped
- moveit_servo.TestGTestProcessPostShutdown test_gtest_pass
  <<< failure message
    Traceback (most recent call last):
      File "/home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/moveit_servo/tests/launch/servo_cpp_integration.test.py", line 133, in test_gtest_pass
        launch_testing.asserts.assertExitCodes(proc_info, process=servo_gtest)
      File "/opt/ros/jazzy/lib/python3.12/site-packages/launch_testing/asserts/assert_exit_codes.py", line 62, in assertExitCodes
        assert info.returncode in allowable_exit_codes, 'Proc {} exited with code {}'.format(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    AssertionError: Proc moveit_servo_cpp_integration_test-5 exited with code 1
  >>>
```

sea-bass, 2024-10-30:
Thanks @mikeferguson -- the actual error is a bit farther up, and I still have no idea what causes it.

```
/home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_ros/moveit_servo/tests/test_integration.cpp:141: Failure
    [moveit_servo_cpp_integration_test-5] Expected equality of these values:
    [moveit_servo_cpp_integration_test-5]   status_curr
    [moveit_servo_cpp_integration_test-5]     Which is: 1-byte object <05>
    [moveit_servo_cpp_integration_test-5]   moveit_servo::StatusCode::NO_WARNING
    [moveit_servo_cpp_integration_test-5]     Which is: 1-byte object <00>
    [moveit_servo_cpp_integration_test-5]
[moveit_servo_cpp_integration_test-5] [  FAILED  ] ServoCppFixture.PoseTest (170 ms)
```

According to the enums in https://github.com/moveit/moveit_msgs/blob/ros2/msg/ServoStatus.msg, it apparently is moving the arm into collision.

It could be that the whole bringup is broken, so the robot starts at zero configuration (which is in collision), sporadically. I sometimes see this happening when I run some of the MoveIt demos locally.

PR #3056 (2024-11-04): Wait for nonzero joint states in PSM in Servo CPP integration test
### Description

I finally chased down the source of the flakiness of the servo integration tests in #3005... I think.

Turns out that the collision monitor thread gets the robot state with the following due to https://github.com/moveit/moveit2/pull/2747 / https://github.com/moveit/moveit2/issues/2748:

```cpp
robot_state_ = planning_scene_monitor_->getPlanningScene()->getCurrentState();
```

However, the Servo integration CPP tests were using

```cpp
/// Helper function to get the current pose of a specified frame.
Eigen::Isom [... cut]
closed 2024-11-04 by sjahr