moveit/moveit2 #3012: Request for a new release of MoveIt into Humble

Danilrivero, 2024-09-25:
### Is your feature request related to a problem? Please describe.

Following up with the release of https://github.com/moveit/moveit2/issues/2928 and https://github.com/moveit/moveit2/pull/3002 to main (aka Jazzy) I have checked the [Release Process](https://moveit.ai/documentation/contributing/releases/) for releases to older versions but could not resolve the cadence for these.

### Describe the solution you'd like

The last Release for Humble dates back to a year ago with the release of [2.5.5](https://github.com/moveit/moveit2/commit/76296af31907b56cb79e1cedb66e80cfb0b84ed7). There are many fixes that could benefit those using this version right now, which is our case with our Stack.

### Describe alternatives you've considered

If the only version being supported for new binaries is Jazzy and there are no plans for Humble it would push us to use the Source version until we migrate to this latest version.

### Additional context

Thanks to everyone involved once again.

rr-tom-noble, 2024-11-11:
@sea-bass @sjahr Do you know how to kickstart this process? It seems like it needs someone with write access to the repos?

Danilrivero, 2024-11-11:
ROS 2 releases that are still being [supported](https://docs.ros.org/en/rolling/Releases.html) should see some sort of cadence as well even though they would not have the frequency of the newest release, I don't know if this is defined somewhere.

The release process for Humble would definitely require some work for the changelogs, it's been a while since the last release

sea-bass, 2024-11-11:
The whole process is listed here:
https://docs.ros.org/en/humble/How-To-Guides/Releasing/Subsequent-Releases.html

Anyone can auto generate a changelog and release locally with this process, but the generation of a tag and then submitting the PR to rosdistro does require being part of the release organization -- which I and others can help with.

The other note is because Humble is not the main version, this should specifically be a patch release (2.5.6).

rr-tom-noble, 2024-11-13:
I'm happy to give this a go once the humble backports for the issues I mentioned are merged. Hopefully in the next couple of days. @sea-bass would it be worth putting out some sort of notification in case there are other soon-to-merge PRs people want to include?

sea-bass, 2024-11-13:
Thanks @rr-tom-noble -- I'll take care of the release once all these backports/fixes are in.

Worth asking since you're bringing a lot of great changes recently: do YOU have any more backports lined up before we cut one?

sea-bass, 2024-11-13:
Oh, it wouldn't happen to be this fix I made last year, would it?

https://github.com/moveit/moveit2/pull/2549

Would be nice if that just needs a backport.

rr-tom-noble, 2024-11-13:
Ah I'm not sure. @rr-mark @rr-aiden ?

rr-mark, 2024-11-13:
@sea-bass That looks like some of what we need, thanks.

We're also trying to recolour attached collision objects, and make collision objects (including attached collision objects) transparent/translucent.

Your PR certainly looks like the right bit of the code for us to look at.

sjahr, 2024-11-14:
@rr-tom-noble @rr-mark Thanks for all your contributions here recently. They are much appreciated! It would be nice to get in touch and sync our collaboration on MoveIt more so we can help you to get your changes even faster into the code base. Would you be available to join the next public maintainer meeting? We're going to invite publicly on discourse but I wanted to notify you earlier so you might be able to block it in your calendar if you're interested

MoveIt Maintainer Meeting
Thursday, November 28 · 5:00 – 6:00pm
Time zone: Europe/Berlin
Google Meet joining info
Video call link: https://meet.google.com/fpo-srqg-feg
Or dial: ‪(US) +1 347-486-5750‬ PIN: ‪644 991 049‬#

rr-tom-noble, 2024-11-14:
@sjahr Thanks for the invite! I'll put it in my calendar :+1:

rr-mark, 2024-11-14:
@sjahr Yes, that would be good.

sea-bass, 2024-11-15:
I don't think it's a good idea to mix binaries and source builds.

If you want to test out your updated branches, you should do something to the effect of `sudo apt remove ros-humble-moveit*`

The problem is because of how apt works, you may need to stick a few more repos like `moveit_msgs` and `moveit_resources` in your workspace to prevent rosdep from reinstalling MoveIt binary packages.

If things work that way, should be good. You can refer to the `.repos` file in the `moveit2_tutorials` repo for what you may need.

---

Also re:  regular cadence for Humble releases, it's a good idea, but as you've seen the active maintainer team is a bit thin right now. I don't even use MoveIt currently, but I'm helping here and there. I'll at least cut the next release soon.

rr-tom-noble, 2024-11-15:
Understood and no worries about the release cadence. Apologies for pushing extra stuff into your workload recently. I greatly appreciate the support you've provided, and that you're helping to manage this release 🙂

sea-bass, 2024-11-17:
The release is good to go, but when I do `bloom-release` I get a permissions error, despite [being in the release team](https://github.com/ros2-gbp/ros2-gbp-github-org/blob/7245c8af5c0e5c856e630bf047f7ee5cef0b387f/moveit.tf#L8).

```
$ bloom-release --rosdistro humble moveit

ROS Distro index file associate with commit '5182bb14955c334a5785d5b591b99d19f2924c0a'
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/5182bb14955c334a5785d5b591b99d19f2924c0a/index-v4.yaml'
==> Fetching 'moveit' repository from 'https://github.com/ros2-gbp/moveit2-release.git'
Cloning into '/tmp/tmpiccze4br'...
remote: Enumerating objects: 140563, done.
remote: Counting objects: 100% (4997/4997), done.
remote: Compressing objects: 100% (987/987), done.
remote: Total 140563 (delta 4454), reused 4010 (delta 4010), pack-reused 135566 (from 1)
Receiving objects: 100% (140563/140563), 46.36 MiB | 19.14 MiB/s, done.
Resolving deltas: 100% (42016/42016), done.
==> Setting release repository remote url to 'git@github.com:ros2-gbp/moveit2-release.git'
==> git remote set-url origin git@github.com:ros2-gbp/moveit2-release.git
==> Testing for push permission on release repository
==> git remote -v
origin  git@github.com:ros2-gbp/moveit2-release.git (fetch)
origin  git@github.com:ros2-gbp/moveit2-release.git (push)
==> git push origin master --dry-run
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Cannot push to remote release repository.
Hint: If you just typed in your username/password and you have two-factor authentication,see:
  http://wiki.ros.org/bloom/Tutorials/GithubManualAuthorization
```

TSNoble, 2024-11-18:
Thanks for all of you work @sea-bass. Given the maintainer team is thin, I'd be happy to try leading future releases (at least until we upgrade to jazzy, and up to the point where write access is needed) since we're using humble quite a lot now, and the binaries are so convenient for us.

I still have an outstanding bug to work on, and I'm feeling like refactoring parts of the pilz code in the future so it's easier to maintain, So I may be pushing for 2.5.7 soon(ish) 😛

rr-tom-noble, 2024-11-18:
Also @sea-bass do you know what ssh key you were using and (if the commands were run inside a docker container, were they mounted)?

sea-bass, 2024-11-18:
> Also [@sea-bass](https://github.com/sea-bass) do you know what ssh key you were using and (if the commands were run inside a docker container, were they mounted)?

I have a key generated for Bloom, used in my `~/.config/bloom`, and running locally, so no mounting issues.

Moreover, I've run `bloom-release` on other packages such as `generate_parameter_library` and `pick_ik` before, which are part of a different PickNik release team that I am also part of.

So not sure. Might legitimately be a permissions issue.

sea-bass, 2024-11-18:
Nah, you were right @rr-tom-noble -- I had SSH keys set up for my old work laptop, but never did it for my personal one 🤦🏻 

PR to `rosdistro` is now queued up with the release!

https://github.com/ros/rosdistro/pull/43534

rr-tom-noble, 2024-11-24:
@sea-bass 

Shall we close this issue out now, and raise new issues as-and-when we need them? (I personally have my eye on #3118, but feels a bit extreme to push for a release for just one commit 😅 ) @Danilrivero would you also be happy with this?

I'm wondering if it'd also be possible to create some kind of pinned issue for Humble releases, which can be updated with links to the individual issues like this one (and future ones). It'd be good for tracking the status of humble, I think, and easier to find where open discussions about the latest release are (or if there is one at all) -- this one was already quite far down the list when I happened to find it

sea-bass, 2024-11-24:
Oh yes, I'll close this.

Re: your suggestion, we may be able to use GitHub discussions for that. Let me make something.

closed 2024-11-24 by sea-bass