moveit/moveit2 #3013: moveit2 planGraspsAndPick function

frankchan12138, 2024-09-26:
### Is your feature request related to a problem? Please describe.

I am an beginner of moveit2, I found that planGraspsAndPick function of MoveGroupInterface is commented. Is there a plan to develop this function?

### Describe the solution you'd like

planGraspsAndPick

### Describe alternatives you've considered

Nothing

mikeferguson, 2024-10-10:
The full discussion is here: https://github.com/moveit/moveit2/issues/1038 - but basically the manipulation code has been deprecated and removed from move_group - you should check out the MTC tutorial which is the recommended alternative: https://moveit.picknik.ai/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html

mikeferguson, 2024-10-10:
PR to remove the dead code: https://github.com/moveit/moveit2/pull/3025

closed 2024-10-10 by mikeferguson