moveit/moveit2 #3021: Move group dizing

t-glez, 2024-10-08:
### Description

I am using ROS2 to control a UR5e with the UR_ROS2_Drivers. It works when using the binary release. When using the source built version move group dies.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

Main

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

Code for calling the move_group node

"""
    # Initialize Arguments
    ur_type = LaunchConfiguration("ur_type")
    use_fake_hardware = LaunchConfiguration("use_fake_hardware")
    safety_limits = LaunchConfiguration("safety_limits")
    safety_pos_margin = LaunchConfiguration("safety_pos_margin")
    safety_k_position = LaunchConfiguration("safety_k_position")
    # General arguments
    description_package = LaunchConfiguration("description_package")
    description_file = LaunchConfiguration("description_file")
    _publish_robot_description_semantic = LaunchConfiguration("publish_robot_description_semantic")
    moveit_config_package = LaunchConfiguration("moveit_config_package")
    moveit_joint_limits_file = LaunchConfiguration("moveit_joint_limits_file")
    moveit_config_file = LaunchConfiguration("moveit_config_file")
    warehouse_sqlite_path = LaunchConfiguration("warehouse_sqlite_path")
    prefix = LaunchConfiguration("prefix")
    use_sim_time = LaunchConfiguration("use_sim_time")
    launch_rviz = LaunchConfiguration("launch_rviz")
    launch_servo = LaunchConfiguration("launch_servo")

    joint_limit_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "joint_limits.yaml"]
    )
    kinematics_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "default_kinematics.yaml"]
    )
    physical_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "physical_parameters.yaml"]
    )
    visual_params = PathJoinSubstitution(
        [FindPackageShare(description_package), "config", ur_type, "visual_parameters.yaml"]
    )

    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare(description_package), "urdf", description_file]),
            " ",
            "robot_ip:=xxx.yyy.zzz.www",
            " ",
            "joint_limit_params:=",
[... 289 lines cut ...]
            "have to be updated.",
        )
    )
    declared_arguments.append(
        DeclareLaunchArgument("launch_rviz", default_value="true", description="Launch RViz?")
    )
    declared_arguments.append(
        DeclareLaunchArgument("launch_servo", default_value="true", description="Launch Servo?")
    )
    """ur_dir = get_package_share_directory('ur_moveit_config')
    ur_launch_dir = os.path.join(ur_dir, 'launch')
    declared_arguments.append(
        IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(ur_launch_dir, 'sensor_kinect_pointcloud.launch.xml'))
        )
    )"""
    return LaunchDescription(declared_arguments + [OpaqueFunction(function=launch_setup),])"""

### Expected behavior

Move group should upload the libraries and publish the planning scene

### Actual behavior

`[ERROR] [move_group-12]: process has died [pid 155208, exit code -6, cmd '/home/marwin/mes_ws/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_77ryurbf --params-file /tmp/launch_params_fkgxrldg --params-file /tmp/launch_params_30pbkqaw --params-file /home/marwin/mes_ws/install/ur_moveit_config/share/ur_moveit_config/config/kinematics.yaml --params-file /tmp/launch_params_7u2wol31 --params-file /tmp/launch_params_h6lpmb1w --params-file /tmp/launch_params_xuu8x1_1 --params-file /tmp/launch_params_cu779mx1 --params-file /tmp/launch_params_siz37d0s --params-file /tmp/launch_params_pn40jguh --params-file /tmp/launch_params_syqj3p3w'].`

t-glez, 2024-10-08:
[launch.log](https://github.com/user-attachments/files/17293419/launch.log)

mikeferguson, 2024-10-29:
I would suggest uninstalling the moveit debians, and rebuilding your workspace. There are some ABI incompatibilities between what is in debs and what is in source and that definitely leads to some of these stranger crashes

closed 2024-11-27 by mikeferguson