moveit/moveit2 #3022: Build error : 1 package failed: srdfdom

minseo104, 2024-10-08:
### Description

Cannot build moveit2

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### If source, which branch?

Humble

### Which RMW are you using?

None

### Steps to Reproduce

I follow this process (https://github.com/moveit/moveit2_tutorials/) and type : $ colcon build --mixin release

### Expected behavior

I also did type : $sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $humble -y , but build failed.

### Actual behavior

[image]
Help me plz. I spend lots of time to solve this problem...m.m

TSNoble, 2024-11-18:
There's a [related issue in setuptools](https://github.com/pypa/setuptools/issues/4483), which may be of some help

rhaschke, 2024-11-18:
Looks like you are not using the system version of setuptools, but /home/user/.local/lib/python3.10/site-packages/setuptools.

closed 2024-11-27 by mikeferguson