moveit/moveit2 #3037: Move multiple arms simultaneously

BingPeng0, 2024-10-18:
Hi all, I managed to control two arms together in a single file using moveit_py with separate controllers: left_arm_controller and right_arm_controller. However, the execution currently happens one arm after the other. I wanted to ask if it is possible to move both arms simultaneously. If so, how can this be done? Thank you.

BingPeng0, 2024-10-18:
I tried to use threading to implement both execution together but it failed. 2 Scenarios will occur.

### 1: Only One Arm is Executed
#### Only One Arm is Executed
[dual_arm_panda.py-1] [INFO] [1729238942.120851031] [moveit_py]: Started Simultaneous Execution
[dual_arm_panda.py-1] [INFO] [1729238942.121780532] [moveit_py]: Planning arm for left side
[dual_arm_panda.py-1] [INFO] [1729238942.121986106] [moveit_py]: Calling PlanningRequestAdapter 'ResolveConstraintFrames'
[dual_arm_panda.py-1] [INFO] [1729238942.122017358] [moveit_py]: Calling PlanningRequestAdapter 'ValidateWorkspaceBounds'
[dual_arm_panda.py-1] [WARN] [1729238942.122020888] [moveit_2775093837.moveit.ros.validate_workspace_bounds]: It looks like the planning volume was not specified. Using default values.
[dual_arm_panda.py-1] [INFO] [1729238942.122028849] [moveit_py]: Calling PlanningRequestAdapter 'CheckStartStateBounds'
[dual_arm_panda.py-1] [INFO] [1729238942.122056025] [moveit_py]: Calling PlanningRequestAdapter 'CheckStartStateCollision'
[dual_arm_panda.py-1] [WARN] [1729238942.122105781] [moveit_2775093837.moveit.planners.ompl.planning_context_manager]: Cannot find planning configuration for group 'left_panda_arm' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[dual_arm_panda.py-1] [INFO] [1729238942.122257637] [moveit_2775093837.moveit.planners.ompl.model_based_planning_context]: Planner configuration 'left_panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[dual_arm_panda.py-1] [INFO] [1729238942.122303547] [moveit_py]: Calling Planner 'OMPL'
[dual_arm_panda.py-1] [INFO] [1729238942.125266114] [moveit_py]: Calling PlanningResponseAdapter 'AddTimeOptimalParameterization'
[dual_arm_panda.py-1] [INFO] [1729238942.125819702] [moveit_py]: Calling PlanningResponseAdapter 'ValidateSolution'
[dual_arm_panda.py-1] [INFO] [1729238942.126051673] [moveit_py]: Calling PlanningResponseAdapter 'DisplayMotionPath'
[dual_arm_panda.py-1] [INFO] [1729238942.126177857] [moveit_py]: Planning arm for right side
[dual_arm_panda.py-1] [INFO] [1729238942.126326132] [moveit_py]: Calling PlanningRequestAdapter 'ResolveConstraintFrames'
[dual_arm_panda.py-1] [INFO] [1729238942.126349842] [moveit_py]: Calling PlanningRequestAdapter 'ValidateWorkspaceBounds'
[dual_arm_panda.py-1] [WARN] [1729238942.126353447] [moveit_2775093837.moveit.ros.validate_workspace_bounds]: It looks like the planning volume was not specified. Using default values.
[dual_arm_panda.py-1] [INFO] [1729238942.126360933] [moveit_py]: Calling PlanningRequestAdapter 'CheckStartStateBounds'
[dual_arm_panda.py-1] [INFO] [1729238942.126372253] [moveit_py]: Calling PlanningRequestAdapter 'CheckStartStateCollision'
[dual_arm_panda.py-1] [WARN] [1729238942.126416247] [moveit_2775093837.moveit.planners.ompl.planning_context_manager]: Cannot find planning configuration for group 'right_panda_arm' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[dual_arm_panda.py-1] [INFO] [1729238942.126526303] [moveit_2775093837.moveit.planners.ompl.model_based_planning_context]: Planner configuration 'right_panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[dual_arm_panda.py-1] [INFO] [1729238942.126571585] [moveit_py]: Calling Planner 'OMPL'
[dual_arm_panda.py-1] [INFO] [1729238942.139903526] [moveit_py]: Calling PlanningResponseAdapter 'AddTimeOptimalParameterization'
[dual_arm_panda.py-1] [INFO] [1729238942.140482845] [moveit_py]: Calling PlanningResponseAdapter 'ValidateSolution'
[dual_arm_panda.py-1] [INFO] [1729238942.140710871] [moveit_py]: Calling PlanningResponseAdapter 'DisplayMotionPath'
[dual_arm_panda.py-1] [INFO] [1729238942.141469484] [moveit_py]: Executing plan
[dual_arm_panda.py-1] [INFO] [1729238942.141540307] [moveit_py]: Executing plan
[dual_arm_panda.py-1] [ERROR] [1729238942.141571649] [moveit_2775093837.moveit.ros.moveit_cpp]: Execution failed! No active controllers configured for group 'right_panda_arm'
[dual_arm_panda.py-1] [INFO] [1729238942.141922272] [moveit_2775093837.moveit.ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[dual_arm_panda.py-1] [INFO] [1729238942.150875043] [moveit_2775093837.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[dual_arm_panda.py-1] [INFO] [1729238942.151043830] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to /left_arm_controller
[ros2_control_node-7] [INFO] [1729238942.151245949] [left_arm_controller]: Received new action goal
[ros2_control_node-7] [INFO] [1729238942.151298186] [left_arm_controller]: Accepted new action goal
[dual_arm_panda.py-1] [INFO] [1729238942.151778950] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: /left_arm_controller started execution
[dual_arm_panda.py-1] [INFO] [1729238942.151801488] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-7] [INFO] [1729238943.910635701] [left_arm_controller]: Goal reached, success!
[dual_arm_panda.py-1] [INFO] [1729238943.951772400] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller '/left_arm_controller' successfully finished
[... cut ...]
[dual_arm_panda.py-1] [INFO] [1729239065.191448356] [moveit_2775093837.moveit.ros.trajectory_execution_manager]: Completed trajectory execution with status PREEMPTED ...
[dual_arm_panda.py-1] [INFO] [1729239065.191532762] [moveit_2775093837.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[ERROR] [dual_arm_panda.py-1]: process has died [pid 67358, exit code -11, cmd '/root/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/dual_arm_panda.py --ros-args -r __node:=moveit_py --params-file /tmp/launch_params_t0abzwke'].

radhen, 2024-10-18:
@BingPeng0 one option is to have a single planning group for both the arms. Suppose you have two two 6 DOFs arms, then create a single planning group with all the 12 DOFs in your URDF/SRDF file for the robot, [in this manner](https://answers.ros.org/question/370136/planning-for-a-dual-arm-robot-in-moveit/). This will allow to set a joint angle goal for all the 12 DOFs, and execute the planned motion such that both the arms move simultaneously, but note you can only plan in joint space with this approach.

BingPeng0, 2024-10-22:
Thanks for your help. I tried out your method and It works! Thank you very much!
Have taken your suggestion and update the first comment accordingly.
Once again. really appreciate your help

closed 2024-10-22 by BingPeng0