moveit/moveit2 #3065: Jerky motion using MoveIt Servo

AleTarsi, 2024-11-06:
### Is your feature request related to a problem? Please describe.

I am experiencing a problem related to the acceleration profile generated by `MoveIt Servo` trajectory planner on my `UR5e`. I’m trying [this example](https://github.com/moveit/moveit2/blob/humble/moveit_ros/moveit_servo/src/cpp_interface_demo/pose_tracking_demo.cpp) of `pose tracking`. For a `bin-picking` scenario.

The documentation specifies that `MoveIt Servo` can enforce only joint position and velocity limits. However, I am experiencing a rough acceleration profile at the start of the motion as shown in the following picture:

[image]

At the starting point, the robot joints make an annoying noise, which I want to avoid.  During the deceleration phase this noise is not present and the deceleration profile seems much smoother.

### Describe the solution you'd like

I have seen that the `time parametrization` in `MoveIt2` uses the [TOTG](https://moveit.picknik.ai/humble/doc/examples/time_parameterization/time_parameterization_tutorial.html#time-parameterization-algorithms) algorithm to define the velocity and acceleration profiles, which enforce up to the joint acceleration limit. However, it seems that the acceleration limit is not read in `MoveIt Servo`. I assume that `TOTG` is used any time you plan using `OMPL` and `Pilz` planner, but I haven’t found this information written somewhere. And I would like to understand if changing to `Pilz` for instance may be a solution to my problem as it enforce at least the acceleration limit.

### Describe alternatives you've considered

I have seen there is the possibility to perform an additional post-processing step and run `[Ruckig](https://moveit.picknik.ai/humble/doc/examples/time_parameterization/time_parameterization_tutorial.html#time-parameterization-algorithms)` algorithm to enforce also the jerk. From the documentation I have seen that this additional plugin is specified in the `ompl_planning.yaml`, therefore I believe it can be used only when planning with OMPL. Is this true?

As my goal is to remove this unwanted sound when I move my robot, do you think that implementing a vanilla `Pilz` planner enforcing the acceleration limits is enough? Do you have examples I can take to do so? I am using ros2 Humble.

Is there the possibility of implementing a `minimum jerk` time parametrization?

Thanks.

AndyZe, 2024-11-06:
Your best bet is probably to change this line: 

https://github.com/moveit/moveit2/blob/f9b5a11a9fbc5e76ec2cc81fbbbb2ad2b70ffc83/moveit_ros/moveit_servo/config/panda_simulated_config_pose_tracking.yaml#L31

To: 

online_signal_smoothing::RuckigFilterPlugin

Which is not in Humble, only later distros. Sorry. I guess you could make a PR to port that change back to Humble.

AndyZe, 2024-11-06:
See https://github.com/moveit/moveit2/blob/main/moveit_core/online_signal_smoothing/src/ruckig_filter.cpp

AleTarsi, 2024-11-12:
Hi @AndyZe! Thank you for your answer. I tested the `Pilz` planner, which significantly reduces the `jerk` at the starting point, enforcing at least the acceleration limit. The resulting motion is visibly smoother and the acceleration peaks are limited as shown in the following figures:

#### Pilz Planner

[image]

#### Servo Planner
[image]

The graph shows the end effector position and the velocity and acceleration value computed through the central difference derivative, I will have a look directly at the joint trajectory, but it seems that this solution works well for my use case.

AndyZe, 2024-11-13:
Happy that you got it working!

closed 2024-11-13 by AndyZe