moveit/moveit2 #3078: Iron Tutorial builds are consistently failing in robotiq_driver tests

TSNoble, 2024-11-10:
### Description

e.g. https://github.com/moveit/moveit2/actions/runs/11748057912/job/32731324687

May be related to https://github.com/PickNikRobotics/ros2_robotiq_gripper/issues/60

The fix there seemed to be that a user was trying to build for one version of ros, but was pulling in a different version of robotiq_driver, leading to errors due to the interface changing.

I'm a little confused as to why this is an issue on the tutorial workflows, since it looks like all of them [use the main version of robotiq_driver](https://github.com/moveit/moveit2_tutorials/blob/59b03cc68ae19cb142be1f0ffc4e19ea800e30d6/moveit2_tutorials.repos#L18-L21), and the tests in question are internal to that repo, so I'd expect both the source and test code to be using the same versions.

### ROS Distro

Rolling

### OS and version

Github Workflows

### Source or binary build?

Source

### If binary, which release version?

N/A

### If source, which branch?

All (as far as I can tell)

### Which RMW are you using?

None

### Steps to Reproduce

Raise a pull request in this repo and wait for the workflow to fail

### Expected behavior

When making moveit changes, these tests should pass

### Actual behavior

When making any changes, the pipeline consistently fails for reasons unrelated to a PR

TSNoble, 2024-11-10:
I think the issue is that the moveit2 tutorial docker image is based on the moveit2 source image, which appears to pull in an iron-compatible iron version of robotiq_driver for the iron builds. The tutorial docker image then pulls in the main version of robotiq_driver via rosdep, and the main tests attempt to run against the iron source.

rr-tom-noble, 2024-11-11:
I tried downloading moveit:moveit2:iron-source, but couldn't find any versions of `robotiq_driver` under `/opt/ros/iron`. That being said, the build was from 4 months ago (presumably when the pipeline started failing). I wonder if that image has _started_ pulling in a `robotiq_driver` binary, hence the failures from that point onwards?

rr-tom-noble, 2024-11-11:
Additionally, I'm unsure if it even makes sense to be running tests for build-from-source dependent packages of the tutorial image, since they're not covered by moveit code.

Perhaps a potential fix would be to set `--cmake-args -DBUILD_TESTING=OFF` for dependencies?

rr-tom-noble, 2024-11-11:
@sea-bass I think it'd make sense to close this given your changes removing the jobs

sea-bass, 2024-11-11:
Closing by ignoring the problem (i.e., not building Iron 😆)

closed 2024-11-11 by sea-bass