moveit/moveit2 #3092: SRDF disable collision link validity checks

rr-tom-noble, 2024-11-15:
### Is your feature request related to a problem? Please describe.

I made a mistake in our SRDF some time ago by disabling collisions using the corresponding tag with a link name that contained a typo. The SRDF loaded fine and (although I didn't get a chance to test collisions) I imagine the ACO will not have reflected the expected configuration.

### Describe the solution you'd like

It seems like we could catch these issues when parsing the SRDF by checking that all links mentioned have already been defined in the URDF. This could either throw an error or, if that seems to extreme, raise a warning that there will be unintended effects.

### Describe alternatives you've considered

Proofreading my code better I suppose 😆

rhaschke, 2024-11-15:
It is desired that you can list objects in the ACO, which are not yet existing as links or collision objects. Thus, at most, an unknown name should raise a warning.

rr-tom-noble, 2024-11-15:
@rhaschke Thanks for confirming. Given it can represent intended behaviour maybe even just an info would be appropriate?

Something like `Disabling collisions with <link> which does not yet exist. If you are not expecting to add this to the scene, there may be a typo`?

TSNoble, 2024-11-16:
Oh actually it looks like these warnings [already exist](https://github.com/TSNoble/srdfdom/blob/148991adf4b3519f3f642f54a13423ab46e7e703/src/model.cpp#L587-L596). This issue can probably be closed.

TSNoble, 2024-11-20:
@rhaschke Sorry to double ping you, but I think this can be closed. I think I was just not observant in noticing the existing warning

closed 2024-12-17 by sea-bass