moveit/moveit2 #3110: SonarScan pipelines are failing on main

TSNoble, 2024-11-17:
### Description

See e.g. https://github.com/moveit/moveit2/actions/runs/11868931947/job/33078707644

I'd be happy to raise a fix (green jobs make me happy), but I'm not too familiar with SonarScan, so some guidance on whether `--ignore-errors mismatch` is the fix, or whether this requires deeper investigation would be appreciated!

### ROS Distro

Humble

### OS and version

SonarScan CI workflow

### Source or binary build?

Source

### If source, which branch?

rolling

### Which RMW are you using?

None

### Steps to Reproduce

Run a SonarScan job

### Expected behavior

Presumably these should not fail

### Actual behavior

They do

### Backtrace or Console output

```
Capture coverage info
  geninfo: WARNING: /usr/include/c++/13/bits/basic_string.h:181: unexecuted block on non-branch line with non-zero hit count.  Use "geninfo --rc geninfo_unexecuted_blocks=1 to set count to zero.
  geninfo: ERROR: mismatched end line for _ZN56RobotTrajectoryTestFixture_ModifyFirstWaypointByPtr_Test8TestBodyEv at /home/runner/work/moveit2/moveit2/.work/target_ws/src/moveit2/moveit_core/robot_trajectory/test/test_robot_trajectory.cpp:361: 361 -> 366
  	(use "geninfo --ignore-errors mismatch ..." to bypass this error)
  Error: Process completed with exit code 1.
```

rhaschke, 2024-11-18:
Will have a look.

rhaschke, 2024-11-18:
Fixes https://github.com/moveit/moveit2/issues/2866

PR #3115 (2024-11-19): CI: Silent lcov errors
Attempt to fix #3110 and #2866.
closed 2024-11-19 by sea-bass