moveit/moveit2 #3114: Inverse kinematics solution with additional custom joint constraints.

aizmailovs, 2024-11-18:
### Is your feature request related to a problem? Please describe.

When using [setFromIK](https://github.com/moveit/moveit2/blob/c2b0e3d6c27ba2f2c7780b4087a5eecc0e600755/moveit_core/robot_state/src/robot_state.cpp#L1446C1-L1449C86), I am receiving an inverse kinematics solution that respects joint limits and avoids collision with objects in the robot's workspace, which is good. 
However, I cannot limit the solution space to additional custom-made joint constraints (`moveit_msgs::msg::JointConstraint`), at least I have not found the way to add them yet.

### Describe the solution you'd like

I would be happy if I could pass additional parameter (array of custom joint constraints) to `setFromIK` that would further limit joint solution space, in addition to joint limits.

### Describe alternatives you've considered

I considered looking for IK solution and then check whether it satisfies custom joint constraints manually. However, it is time costly.  

When looking into [setFromIK](https://github.com/moveit/moveit2/blob/c2b0e3d6c27ba2f2c7780b4087a5eecc0e600755/moveit_core/robot_state/src/robot_state.cpp#L1446C1-L1449C86), I saw that there is a `const GroupStateValidityCallbackFn& constraint` parameter. Is it possible to use it to check the validity of the sampled state including additional joint constraints?

aizmailovs, 2024-11-20:
The [GroupStateValidityCallbackFn](https://github.com/moveit/moveit2/blob/c2b0e3d6c27ba2f2c7780b4087a5eecc0e600755/moveit_core/robot_state/include/moveit/robot_state/robot_state.h#L68-L70) can be used to create custom validation for sampled IK solutions. In my case, I use it to validate solutions against additional joint constraints. For reference, the code is provided below.

The current logic is that the solution is sampled and then validated. 
However, I am curious whether it is possible to directly apply constraints to IK solver. In my case, it would be to update the solution space to exclude joint configurations that are beyond user-defined joint constraints.

Code:
```#include <memory>
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>
#include <thread>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit_msgs/msg/display_trajectory.hpp>
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/robot_state/robot_state.h>

int main(int argc, char *argv[])
{
    // Initialize ROS and create the Node
    rclcpp::init(argc, argv);
    auto const node = std::make_shared<rclcpp::Node>(
        "joint_constraints_node",
        rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true));

    // Create a ROS logger
    auto const logger = rclcpp::get_logger("joint_constraints_node");

    // Spin up a SingleThreadedExecutor
    rclcpp::executors::SingleThreadedExecutor executor;
    executor.add_node(node);
    auto spinner = std::thread([&executor]()
                                { executor.spin(); });  

    const std::string planning_group_name = "<planning_group_name>";

    // Create the MoveIt MoveGroup Interface
    using moveit::planning_interface::MoveGroupInterface;
    auto move_group_interface = MoveGroupInterface(node, planning_group_name);

    robot_model_loader::RobotModelLoader robot_model_loader(node);
    const moveit::core::RobotModelPtr& kinematic_model = robot_model_loader.getModel();

    moveit::core::RobotStatePtr kinematic_state(new moveit::core::RobotState(kinematic_model));
    const moveit::core::JointModelGroup* joint_model_group = kinematic_model->getJointModelGroup(planning_group_name);

    auto current_pose = move_group_interface.getCurrentPose();

    moveit_msgs::msg::Constraints mixed_constraints;

    moveit_msgs::msg::JointConstraint joint1_constraint;

    joint1_constraint.joint_name = "joint_1";
    joint1_constraint.position = 0.0;
    joint1_constraint.tolerance_below = 0.0872665;
    joint1_constraint.tolerance_above = 1.65806;
    joint1_constraint.weight = 0.2;

    mixed_constraints.joint_constraints.emplace_back(joint1_constraint);

    move_group_interface.setPathConstraints(mixed_constraints);

    // Set a target Pose
    auto const target_pose = [&current_pose]
    {
      geometry_msgs::msg::Pose msg;
      msg.orientation = current_pose.pose.orientation;
      msg.position.x = 0.0;
      msg.position.y = 1.125;
      msg.position.z = 0.3;
      return msg;
    }();
[... 33 lines cut ...]
                                                return false; // Reject IK solution
                                            }
                                        }

                                        RCLCPP_INFO(logger, "IK solution accepted.");
                                        return true; // Accept IK solution
                                        });

    if (found_ik)
    {
        std::vector<double> joint_values;
        kinematic_state->copyJointGroupPositions(joint_model_group, joint_values); //joint_values can be used to set joint target for path-planning.

    }
    else
    {
        RCLCPP_ERROR(logger, "No valid IK solution found.");
    }

    // Shutdown ROS
    rclcpp::shutdown(); // <--- This will cause the spin function in the thread to return
    spinner.join();     // <--- Join the thread before exiting
    return 0;

}

rhaschke, 2024-11-21:
Unfortunately, the IK solver API doesn't allow the passing of constraints. Hence, constraints can only be considered via validation and rejection.

aizmailovs, 2024-11-21:
Thank you for your reply @rhaschke! I appreciate the clarification!

I have some additional questions for potential workarounds / enhancements.

1) Is it possible to update API to include joint constraints?
2) Could the available joint space (a combination of joint position limits and joint constraints) be pre-processed and passed to the IK solver API as "updated joint position limits"?

AndyZe, 2024-11-21:
I'd suggest to fork bio_ik and add whatever constraints you need there. That's exactly what I've been doing.

rhaschke, 2024-11-21:
I know that bio_ik in principle can handle constraints. However, you need to call the service directly, don't you?
MoveIt's plugin API for IK solvers doesn't allow to pass constraints:
https://github.com/moveit/moveit2/blob/648724955963d8d77d16178a2f951f58fbb69986/moveit_core/kinematics_base/include/moveit/kinematics_base/kinematics_base.h#L174-L176

AndyZe, 2024-11-21:
I modified `bio_ik/src/kinematics_plugin.cpp`

closed 2025-02-26 by github-actions[bot]