moveit/moveit2 #3116: Cut a 2.10.1 release for Jazzy

sea-bass, 2024-11-19:
### Is your feature request related to a problem? Please describe.

Assigning myself this task next.

I'm basically going to branch off before the new Cartesian planner, Ruckig servo smoothing plugin, fuzzy-matching trajectory cache, and other new messages introduced.

### Describe the solution you'd like

Jazzy has an updated release with fixes!

### Describe alternatives you've considered

Having a life

mikeferguson, 2024-11-19:
Given that people are just now starting to really use Jazzy - perhaps just branch as-is at the moment? We've always had a bit of a lag after a new distro initially goes out where we were OK with API/ABI changes, although I don't know if we have codified that anywhere (and we are basically right now at the 6 month mark from first release of Jazzy).

Unfortunately, I don't think we have a good way to determine who / how many are actually depending on Jazzy API stability yet (at ROSCon, it appeared about 5-10% of users had maybe moved to Jazzy in the ROS community, but I can't tell how many of those folks are core ROS devs versus end users... I just know that real robots are slow to move to new distros).

sea-bass, 2024-11-19:
I didn't know that was an option, and would certainly make life easier! Maybe I'll bring it up at the next maintainer meeting to get consensus.

sea-bass, 2024-11-28:
Based on the maintainer meeting, we're going to merge #3113 and then release a 2.12 for Jazzy and Rolling.

This will also mark the branch-off point of a new `jazzy` branch as well.

closed 2024-12-17 by sea-bass