moveit/moveit2 #3117: sequence_move_group Unexpectedly Returns UNKNOWN Code

maxwelllls, 2024-11-19:
### Description

When testing the `sequence_move_group `functionality of Pilz for sequential trajectory motion planning, this bug is triggered after running it only a few times.

After sending an `async_send_goal() `request, the result_callback is immediately invoked, and the returned result.code is UNKNOWN.

The provided code is a simplified minimal reproducible scenario. During actual testing, this issue occurs with both single-segment and multi-segment trajectories.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

CycloneDDS

### Steps to Reproduce

1. Configure the pilz planning library and launch the robotic arm package.
2. Run the test code `ros2 run sequence_move_demo demo`

NOTE:
Change the "master_arm" in the code to your robot move group,and set joint at 'home' pose
```
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit_msgs/action/move_group_sequence.hpp>
#include <rclcpp_action/rclcpp_action.hpp>
#include <moveit/robot_model_loader/robot_model_loader.h>
#include <moveit/kinematic_constraints/utils.h>
#include <rapidjson/document.h>
#include <chrono>
#include <memory>
#include <string>
#include <map>
#include <vector>
#include <thread>
#include <functional>
#include <cmath>

class SequenceMoveDemo : public rclcpp::Node
{
public:
    SequenceMoveDemo();
    ~SequenceMoveDemo();
    void initialize();
    void run();

private:
    // Member variables
    moveit::core::RobotModelPtr moveit_robot_model_;
    std::shared_ptr<rclcpp_action::Client<moveit_msgs::action::MoveGroupSequence>> sequence_move_group_client_;
    moveit_msgs::msg::MotionSequenceRequest sequence_request_;
    bool last_trajectory_connect_bool_;
    std::thread worker_thread_;
    std::vector<double> last_joint_positions_; // Store last joint positions
};

SequenceMoveDemo::SequenceMoveDemo()
    : Node("sequence_move_demo"), last_trajectory_connect_bool_(false)
{
    // Initialize the action client
[... 236 lines cut ...]
[INFO] [1732012231.783669544] [sequence_move_demo]: Number of items in request: 1
[INFO] [1732012231.783699079] [sequence_move_demo]: Sending goal with plan_only=false
[INFO] [1732012231.784493893] [sequence_move_demo]: Planning request accepted
[INFO] [1732012232.443191784] [sequence_move_demo]: Planning request result code= 4
[INFO] [1732012232.443242262] [sequence_move_demo]: Error code: 1
[INFO] [1732012232.443257714] [sequence_move_demo]: Planning time: 0.000000 seconds
[INFO] [1732012232.443322554] [sequence_move_demo]: Planning request complete!
[INFO] [1732012232.448162755] [sequence_move_demo]: Cleared sequence request items.
[INFO] [1732012237.448327356] [sequence_move_demo]: Set start state to last joint positions.
[INFO] [1732012237.448659886] [sequence_move_demo]: Number of items in request: 1
[INFO] [1732012237.448686806] [sequence_move_demo]: Sending goal with plan_only=false
[INFO] [1732012237.449232138] [sequence_move_demo]: Planning request accepted
[INFO] [1732012238.102866545] [sequence_move_demo]: Planning request result code= 4
[INFO] [1732012238.102929447] [sequence_move_demo]: Error code: 1
[INFO] [1732012238.102946452] [sequence_move_demo]: Planning time: 0.000000 seconds
[INFO] [1732012238.102964450] [sequence_move_demo]: Planning request complete!
[INFO] [1732012238.103834674] [sequence_move_demo]: Cleared sequence request items.
[INFO] [1732012243.104020065] [sequence_move_demo]: Set start state to last joint positions.
[INFO] [1732012243.104996102] [sequence_move_demo]: Number of items in request: 1
[INFO] [1732012243.105047145] [sequence_move_demo]: Sending goal with plan_only=false
[INFO] [1732012243.105816297] [sequence_move_demo]: Planning request accepted
[INFO] [1732012243.106202476] [sequence_move_demo]: Planning request result code= 0
[INFO] [1732012243.106232580] [sequence_move_demo]: Error code: 0
[INFO] [1732012243.106248451] [sequence_move_demo]: Planning time: 0.000000 seconds
[INFO] [1732012243.106264605] [sequence_move_demo]: Planning request unknown result code

maxwelllls, 2024-11-21:
Hi, @sea-bass ,

After a few days of troubleshooting, by comparing the differences between `sequence_move_group `and `move_action`, I have identified that the issue is likely related to **Reentrant** functionality.

When I removed the `action_callback_group_ `parameter in the `create_server() `function, the program was able to run stably, indicating a potential multi-threading race condition.

I also tried adding mutexes to the `goal_handle_ `and `move_state_ `objects, which improved stability, but the issue still occurred after running a stress test for over 20 minutes.

Is the **Reentrant** characteristic of the Pilz planner essential? If it must support **Reentrant**, could you provide further suggestions for troubleshooting this issue?

sea-bass, 2024-11-21:
Interesting... I definitely don't have the full context, but of course reentrant callbacks allow multiple callbacks on this sequence action server to run in parallel.

If Pilz is not thread-safe, I could see how that causes problems.

If instead of removing `action_callback_group_`, you instead initialize it to a `MutuallyExclusiveCallbackGroup`, how does that work with your tests?

It may still be beneficial to keep these callbacks in their own group, just not allow them to cause race conditions.

sea-bass, 2024-11-21:
BTW this seems to have been added pretty early on in the "port MoveIt to ROS 2" days, without a pull request even.

https://github.com/moveit/moveit2/commit/b9239108ed71d03b4dc6edf8994638504bbd105a

So I would imagine this probably can safely be replaced with a mutually exclusive callback group if it fixes the race condition.

@henningkayser tagging you too in case you have inputs.

sea-bass, 2024-11-21:
Yep, that's exactly it! If it works well, definitely put up a PR and tag us.

closed 2024-11-22 by maxwelllls