moveit/moveit2 #3130: Several OMPL interface tests started failing with `humble_ci` job on main

sea-bass, 2024-11-23:
### Description

This only started a few days ago (w.r.t. the issue creation date):
https://github.com/moveit/moveit2/actions/runs/11988362896/job/33423544280

This is unique to the `humble_ci` job on the `main` branch -- we need to either pinpoint and fix the cause of failure, or draw the line to say that the `main` branch no longer works on Humble.

I'd like to see if we can figure out the issue first, and only resort to the latter option if absolutely necessary.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

https://github.com/moveit/moveit2/actions/runs/11988362896/job/33423544280

### Expected behavior

Tests should pass

### Actual behavior

Tests do not pass

rr-tom-noble, 2024-11-25:
Related to https://github.com/moveit/moveit2/issues/1952

It seems to be geometric_shapes throwing the error. Specifically, [these lines](https://github.com/moveit/geometric_shapes/blob/c24eb4940646b4584f8961c75e4807a44637e3d7/src/bodies.cpp#L603-L604)

sea-bass, 2024-11-26:
Ah! Yeah, turns out there was a commit to `geometric_shapes` that landed 5 days ago: https://github.com/moveit/geometric_shapes/pull/239

And this repo is actually building it from source.

I've attempted to pin the SHA of this dependency in https://github.com/moveit/moveit2/pull/3132 to see if this is it...

sea-bass, 2024-11-26:
Found the issue. https://github.com/moveit/geometric_shapes/pull/252 should fix it.

closed 2024-11-26 by sea-bass