moveit/moveit2 #3134: Fix header install destination

rhaschke, 2024-11-27:
### Description

We frequently see bug reports about ABI incompatibilities when overlaying existing binary packages.
I strongly believe that this related to the header installation problem raised in https://github.com/ros2/ros2/issues/1150.
For ROS Humble, the TSC decided to install headers into a subfolder /opt/ros/<distro>/include/<package>/<package>/:
https://docs.ros.org/en/humble/Releases/Release-Humble-Hawksbill.html#c-headers-are-installed-in-a-subdirectory
https://colcon.readthedocs.io/en/released/user/overriding-packages.html#install-headers-to-a-unique-include-directory

MoveIt2 never implemented that change, causing ABI issues with package overlays to persist.
We recently fixed this for srdfdom: https://github.com/moveit/srdfdom/pull/132 and https://github.com/moveit/srdfdom/commit/65ce909735f95d781e22496672bd9a8e35fc6baf.
I think, it's time to go through all MoveIt-releated packages and eventually fix it.

### ROS Distro

Humble

### OS and version

Ubuntu

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

-

### Expected behavior

-

### Actual behavior

-

mikeferguson, 2024-11-27:
@TSNoble you might be interested in working on this - once released, it would avoid many (all?) of the "source build overlaying binaries causes weird ABI crashes" that I think were plaguing you recently.

rhaschke, 2024-11-27:
There is another example what to do in https://github.com/moveit/geometric_shapes/pull/253.

rr-tom-noble, 2024-11-28:
To be fair, I think most of our failures have just been poor dependency management on our part e.g. the last failure was caused by building against 2.5.5, but linking against 2.5.6.

Currently on my agenda:
1. Get https://github.com/moveit/moveit2/pull/3113 merged
2. Pick up follow on header work listed [here](https://github.com/moveit/moveit2/issues/857)
3. Implement the proposal in https://github.com/moveit/moveit2/issues/3060 (@rhaschke unrelated, but would be good to get your thoughts)

This ticket feels like it'd sit quite nicely with the rest of the items in point 2, which all feel fairly small in isolation, so I don't mind taking a shot at it. Also, thanks for the examples. I'll probably work through that list after 1, although work is very busy currently so mainly doing this in my spare time 😅

rahul3002, 2025-01-02:
I want to work on this issue its pretty straight forward

mikeferguson, 2025-01-02:
> I want to work on this issue its pretty straight forward

Please do - open a PR when you're ready!