moveit/moveit2 #3135: CHOMP doesn't respect joint constraints.

aizmailovs, 2024-11-27:
### Description

CHOMP does not respect the joint constraints (`moveit_msgs::msg::JointConstraint`) when planning a motion.

### ROS Distro

Humble

### OS and version

Ubuntu 22.04.

### Source or binary build?

Binary

### If binary, which release version?

2.5.5.

### Which RMW are you using?

None

### Steps to Reproduce

1. Get a working robot arm simulation in Moveit2, created using Moveit Setup Assistant. Doesn't matter which robot in particular is in use. 
If it is possible plan and execute motions using this simulation - the step succeeded.

2. Install CHOMP motion planner `sudo apt-get install ros-humble-moveit-planners-chomp`.

3. Create and add CHOMP config file `chomp_planning.yaml` at `<robot>_moveit_config/config`.
File content in my case:
```
planning_plugin: chomp_interface/CHOMPPlanner
enable_failure_recovery: true
jiggle_fraction: 0.05
request_adapters: >-
    default_planner_request_adapters/AddTimeOptimalParameterization
    default_planner_request_adapters/ResolveConstraintFrames
    default_planner_request_adapters/FixWorkspaceBounds
    default_planner_request_adapters/FixStartStateBounds
    default_planner_request_adapters/FixStartStateCollision
    default_planner_request_adapters/FixStartStatePathConstraints
ridge_factor: 0.01
start_state_max_bounds_error: 0.1
trajectory_initialization_method: "linear"
```

4. Add CHOMP planning pipeline:
```
    moveit_config = (
        MoveItConfigsBuilder("<robot>", package_name="<robot>_moveit_config")
        .planning_pipelines(pipelines=["ompl", "chomp"])
        .to_moveit_configs()
    )
```

5. Create a node to:
- apply joint constraints, 
- set planning goal (in joint-space),
- set CHOMP planning pipeline,
- plan a motion

Here is the code that I used for it:
```
#include <thread>
#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

[... 340 lines cut ...]
[move_group-3] [INFO] [1732712678.751386930] [chomp_planner]: Planned with Chomp Parameters (learning_rate, ridge_factor, planning_time_limit, max_iterations), attempt: # 1 
[move_group-3] [INFO] [1732712678.751391148] [chomp_planner]: Learning rate: 0.010000 ridge factor: 0.010000 planning time limit: 10.000000 max_iterations 200 
[move_group-3] [WARN] [1732712678.751616733] [moveit_trajectory_processing.time_optimal_trajectory_generation]: Joint acceleration limits are not defined. Using the default 1 rad/s^2. You can define acceleration limits in the URDF or joint_limits.yaml.
[move_group-3] [ERROR] [1732712678.769967864] [moveit.ros_planning.planning_pipeline]: Computed path is not valid. Invalid states at index locations: [ 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 ] out of 109. Explanations follow in command line. Contacts are published on /display_contacts
[move_group-3] [INFO] [1732712678.769995246] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.114623, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770018209] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.143131, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770033899] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.171640, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770049158] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.199959, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770064066] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.223781, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770110073] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.241320, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770126654] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.252578, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770141883] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.257578, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770157312] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.257009, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770172411] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.252813, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770187699] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.242480, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770202598] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.225255, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770217315] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.201138, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770231963] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.170865, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770246190] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.139593, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [INFO] [1732712678.770260156] [moveit_kinematic_constraints.kinematic_constraints]: Constraint violated:: Joint name: 'joint_1', actual value: -0.108321, desired value: 0.000000, tolerance_above: 1.658060, tolerance_below: 0.087266
[move_group-3] [ERROR] [1732712678.770268903] [moveit.ros_planning.planning_pipeline]: Completed listing of explanations for invalid states.
[move_group-3] [INFO] [1732712678.770625406] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was found but it seems to be invalid (possibly due to postprocessing). Not executing.
```

Can somebody confirm or deny this behavior, please?

ANIR1604, 2024-12-04:
I would like work on this, if that's okay.

aizmailovs, 2024-12-06:
Hi, @ANIR1604!
I'm glad to hear you're willing to work on this.
Please feel free to share your findings, and don't hesitate to reach out if you have any questions or need assistance reproducing the issue.