moveit/moveit2 #3147: stderr: moveit_setup_sdrf_plugins

Sunpreet29, 2024-12-03:
### Description

While building the moveit workspace, I encountered this issue.

--- stderr: moveit_setup_srdf_plugins
/home/sunny/ws_moveit2/src/moveit2/moveit_setup_assistant/moveit_setup_srdf_plugins/include/moveit_setup_srdf_plugins/rotated_header_view.hpp:0: Note: No relevant classes found. No output generated.
---
Summary: 55 packages finished [58.7s]
  1 package had stderr output: moveit_setup_srdf_plugins

I checked the rotated_header_view.hpp file on the Github page and it has the same code as the one in my system. Is this even a non-critical error that could be ignored?

### ROS Distro

Humble

### OS and version

Ubuntu 22.04 on WSL2 (Windows 11)

### Source or binary build?

Source

### Which RMW are you using?

None

### Steps to Reproduce

Go to the moveit workspace and run the build command.

```
cd ~/ws_moveit_2
colcon build --mixin release
```

### Expected behavior

The build should be clean without an error. The following output should be shown
`finished`

### Actual behavior

Summary: 55 packages finished [58.7s]
  1 package had stderr output: moveit_setup_srdf_plugins

sea-bass, 2024-12-04:
The output says `55 packages finished`, so build was successful.

See https://github.com/colcon/colcon-core/issues/203#issuecomment-512637557 for more information.

closed 2024-12-04 by sea-bass

ryanzhtr, 2024-12-27:
This stderr will not affect your use of most functions in moveit2, if you want to fix this stderr, you can add 'Q_OBJECT' as a macro definition after line 46 in 'rotated_header_view.hpp', this will generate moc_*.cpp files, which contain the implementation functions of signal.