moveit/moveit2 #3157: Installation from source with Ros-Jazzy failed

Pittmann-XIE, 2024-12-11:
### Description

the step "colcon build --mixin release" failed.

PS: why I use ros-jazzy? Because I want to use the python APIs of moveit2. If ros-jazzy is the not the correct option, could you tell me, which version of ros2 I should install in order to use python APIs of moveit2?

### ROS Distro

Jazzy

### OS and version

Distributor ID: Ubuntu Description:    Ubuntu 24.04.1 LTS Release:        24.04 Codename:       noble

### Source or binary build?

Source

### If source, which branch?

main 

### Which RMW are you using?

None

### Steps to Reproduce

1. use the docker image ros-jezzy: osrf/ros:jazzy-desktop-full
2. build the container from the image
3. follow the steps on https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html

### Expected behavior

There should be no error when I run the command "colcon build --mixin release"

### Actual behavior

--

### Backtrace or Console output

Finished <<< rviz_marker_tools [9.19s]                                                                                                                                               
--- stderr: moveit_msgs                                                                                                                                                 
/bin/sh: 1: Syntax error: "(" unexpected
gmake[2]: *** [CMakeFiles/ament_cmake_python_copy_moveit_msgs.dir/build.make:70: CMakeFiles/ament_cmake_python_copy_moveit_msgs] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:452: CMakeFiles/ament_cmake_python_copy_moveit_msgs.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_msgs [11.4s, exited with code 2]
Aborted  <<< kortex_driver [5.78s]                                                                                                                  
Aborted  <<< robotiq_driver [16.0s]                                                                                      
Aborted  <<< rviz_visual_tools [34.2s]                                   

Summary: 16 packages finished [34.4s]
  1 package failed: moveit_msgs
  3 packages aborted: kortex_driver robotiq_driver rviz_visual_tools
  5 packages had stderr output: kortex_api kortex_driver moveit_msgs robotiq_controllers rviz_visual_tools
  51 packages not processed

Pittmann-XIE, 2024-12-11:
update:

### There's also problem with ros-iron
### Steps to reproduce
1. build the container from this image "docker pull osrf/ros:iron-desktop-full"
2. follow the instructions on https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html

### Terminal output
stderr: robotiq_controllers
CMake Error at CMakeLists.txt:10 (find_package):
  By not providing "Findcontroller_interface.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "controller_interface", but CMake did not find one.

  Could not find a package configuration file provided by
  "controller_interface" with any of the following names:

  controller_interfaceConfig.cmake
  controller_interface-config.cmake

  Add the installation prefix of "controller_interface" to CMAKE_PREFIX_PATH
  or set "controller_interface_DIR" to a directory containing one of the
  above files.  If "controller_interface" provides a separate development
  package or SDK, be sure it has been installed.

Failed   <<< robotiq_controllers [0.49s, exited with code 1]
Aborted  <<< robotiq_driver [0.02s]
Aborted  <<< moveit_msgs [0.64s]                                                                                                                                             
Aborted  <<< moveit_configs_utils [0.66s]
Aborted  <<< moveit_resources_panda_moveit_config [0.72s]                                                                                                               
Aborted  <<< kortex_description [0.69s]
Aborted  <<< dual_arm_panda_moveit_config [0.71s]
Aborted  <<< moveit_resources_fanuc_moveit_config [0.74s]
Aborted  <<< rviz_visual_tools [1.47s]                                                                                                       
Aborted  <<< rviz_marker_tools [1.82s]                                                                      
Aborted  <<< kortex_driver [5.86s]                             

Summary: 8 packages finished [6.57s]
  1 package failed: robotiq_controllers
  10 packages aborted: dual_arm_panda_moveit_config kortex_description kortex_driver moveit_configs_utils moveit_msgs moveit_resources_fanuc_moveit_config moveit_resources_panda_moveit_config robotiq_driver rviz_marker_tools rviz_visual_tools
  4 packages had stderr output: kortex_driver moveit_msgs robotiq_controllers rviz_visual_tools
  52 packages not processed

sea-bass, 2024-12-11:
For the Jazzy case, it seems like you were possibly using `sh` instead of `bash` in your Docker container: https://stackoverflow.com/questions/31075438/getting-error-bin-sh-1-syntax-error-unexpected

For the Iron case, you need to ensure that `rosdep`s are installed as shown here: https://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html#rosdep-operation (this is also covered in the tutorials)